dotnet/iot #881: Debug builds don't declare DEBUG define

pgrawehr, 2019-12-06:
**Describe the bug**
If I add the following to a .cs file in the System.Device.Gpio project and build a debug configuration (Linux-Debug or Windows-Debug) the wrong line is ignored:
```
#if DEBUG
                    ThisWontCompile();
#endif
```

**Steps to reproduce**
Add the above to a cs file and compile in debug mode (either in VS 2019 or on the command line using `dotnet build -c "Linux-Debug" System.Device.Gpio.sln`

**Expected behavior**
The build fails with a compile error. 

**Actual behavior**
The build succeeds. 

**Versions used**
Latest master of dotnet/iot

NET SDK Version 3.0.1
```
.NET Core SDK (gemäß "global.json"):
 Version:   3.0.101
 Commit:    bc5f8df0f5

Laufzeitumgebung:
 OS Name:     Windows
 OS Version:  10.0.17763
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\3.0.101\

Host (useful for support):
  Version: 3.1.0
  Commit:  65f04fb6db

.NET Core SDKs installed:
  2.1.202 [C:\Program Files\dotnet\sdk]
  2.1.400 [C:\Program Files\dotnet\sdk]
  2.1.402 [C:\Program Files\dotnet\sdk]
  2.1.502 [C:\Program Files\dotnet\sdk]
  2.1.503 [C:\Program Files\dotnet\sdk]
  2.1.508 [C:\Program Files\dotnet\sdk]
  2.1.509 [C:\Program Files\dotnet\sdk]
  3.0.101 [C:\Program Files\dotnet\sdk]
  3.1.100 [C:\Program Files\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.1.14 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.1.14 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.0 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.0.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.14 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.0.1 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.0 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
```

pgrawehr, 2019-12-06:
One way of solving it is by adding something like 
```
<PropertyGroup Condition="$(Configuration.EndsWith('Debug'))=='true'">
    <DefineConstants>DEBUG;TRACE</DefineConstants>
</PropertyGroup>
```

To Directory.Build.props, but I think that should be part of the default build scripts.

joperezr, 2019-12-09:
I'm not able to repro this, I just did a diagnostic build of System.Device.Gpio and I can see that DEBUG constant is correctly being passed in:

```
F:\git\iot\.dotnet\dotnet.exe exec "C:\Users\joperezr\.nuget\packages\microsoft.net.compilers.toolset\3.1.0-beta1-19127-06\tasks\netcoreapp2.1\bincore\csc.dll" /noconfig /unsafe+ /checked- /nowarn:1701,1702,1705,1591,NU5105,1701,1702 /nostdlib+ /platform:AnyCPU /errorreport:prompt /warn:4 /doc:F:\git\iot\artifacts\obj\System.Device.Gpio\Debug\netstandard2.0\linux\System.Device.Gpio.xml /define:TRACE;DEBUG;NETSTANDARD;NETSTANDARD2_0 /highentropyva+ /reference:C...
```

As you can see, the build is succesfully setting the following defines: `/define:TRACE;DEBUG;NETSTANDARD;NETSTANDARD2_0`

pgrawehr, 2019-12-10:
@joperezr 

- Add the above code snippet to RaspberryPi3Driver.Linux.cs (basically a syntax error in debug mode)
- `cd iot`, then `build`
- This fails, as expected (I didn't check this before)
- `cd src\System.Device.Gpio`, then `dotnet build -c "Linux-Debug" System.Device.Gpio.sln`
- This passes, despite no changes to the code
- Open System.Device.Gpio.sln in Visual Studio, then perform a build in Linux-Debug configuration
- This passes as well

So it seems that when the build executes directly in MsBuild, DEBUG is not defined (which can also be verifed by checking the properties of the projects in Visual Studio).

joperezr, 2019-12-10:
Oh I see, yeah that makes sense. This is happening only because when you either build through the command line or through VS, you are building the Linux-Debug configuration, which for that one we are not defining the specific constant. The Linux-Debug and Windows-Debug configurations are basically just a convenience configuration that were only created so that development with VS would be easier, but that one is not really built when building with build.cmd. VS doesn't support cross-targeting for different RIDs, so we have built custom logic into the repo in order to support this, but there are few things that won't work perfectly like what you have found. I have chatted with @krwq and the idea is to remove the cross-targeting logic and only produce one assembly for both linux and windows, so ideally this won't be a problem in the near future.

Is there a specific reason why you would want to fix this now? Or in other words, is this blocking you?

pgrawehr, 2019-12-11:
With all the low-level dependencies we have, creating just one assembly will be tricky, I think. 
No, this isn't really causing any problems for now. I just came across this when I was using the `Debug.Fail` method in #823 and I got a Resharper warning about this call doing nothing.

joperezr, 2019-12-11:
> With all the low-level dependencies we have, creating just one assembly will be tricky, I think.

It will take a bit of refactoring work, but we have a plan more or less so we should be able to do it for vNext.

>  I was using the Debug.Fail method in #823 and I got a Resharper warning about this call doing nothing.

Oh in this case don't trust in Resharper too much since as I said before, MSBuild SDK doesn't allow cross-rid targeting inbox so we had to write some custom targets for this which might confuse VS a bit. In the end when you build with build.cmd, you should see the right defines getting set, and Debug.Fail call will actually work as expected.

pgrawehr, 2019-12-12:
Well, yes. But I normally build using VS when I want to debug some code. But as said, it's not a big deal right now, since using of Debug.Fail and similar methods isn't the best concept, anyway.

pgrawehr, 2023-03-18:
The problem appears to have been resolved meanwhile. The code in the original post now fails.

closed 2023-03-18 by pgrawehr