dotnet/iot #882: Color Sensor's code not working

ClementG-63, 2019-12-09:
Each mode of the ColorSensorMode is not working except the mode "Color".
RGB, Reflexion, Ambient are giving None and value equals 0.

```csharp
       static void Main(string[] args)
        {
            Brick _brick = new Brick();
            EV3ColorSensor color = new EV3ColorSensor(_brick, SensorPort.Port3);
            color.ColorMode = ColorSensorMode.Ambient;

            int count = 0;
            while (count < 100)
            {
                Console.WriteLine(color.ReadColor());

                var value = color.Read();
                Console.WriteLine(value + "%");

                if (value == 0)
                {
                    color.SelectNextMode();
                    Console.WriteLine("Mode changed" + color.SelectedMode());
                }

                Task.Delay(300).Wait();
            }
        }
```

**Expected behavior**

That's testing each mode, if 0 given : swap mode.
Result : 
None
0%
Mode changedReflection
None
0%
Mode changedGreen
None
0%
Mode changedBlue
None
0%
Mode changedRed
None
0%
Mode changedColor
Brown
7% (not percentage)

Last Version Used
Add following information:

- `dotnet --info` : 
SDK .NET Core (reflétant tous les global.json) :
 Version:   3.0.100
- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
- Version of `System.Device.Gpio` package : Last Version
- Version of `Iot.Device.Bindings` package : Last Version

krwq, 2019-12-09:
cc: @Ellerbach

Ellerbach, 2019-12-10:
@ClementG-63, I'll try to see what's happening. As there are few different color sensors, may I ask you to send me a picture of your module?
Also, when changing color mode, you need to allocate some time to the sensor to read the new value. 
I don't have access to the hardware right now, at best I can test early next week.

Ellerbach, 2019-12-15:
@ClementG-63 , so I tested the code with my own sensor and I get colors.
The EV3 Color Sensor is this one:
[image]
This is what I get on a brown surface (wood) when using the "green" mode (the first one with red light) :
EV3 Color Sensor, Raw: 0, ReadASString: 0
Color: None, Red: 32, Green: 2, Blue: 141

With the next more, the "Blue" one (where light is actually not blue, still red):
EV3 Color Sensor, Raw: 0, ReadASString: 0
Color: None, Red: 77, Green: 0, Blue: 108

This is what I get when using the mode "Color" one (ambient one with the multi color light):
EV3 Color Sensor, Raw: 7, ReadASString: Brown
Color: Brown, Red: 0, Green: 0, Blue: 0
Sometimes, it does detect Yellow (that's normal vs the color of my wood)

And this is what I get when using the ambient and reflection modes (with the blue light):
EV3 Color Sensor, Raw: 0, ReadASString: 0
Color: None, Red: 0, Green: 0, Blue: 0

So there may be an issue with the 2 last modes. But that said comparing with the values from the EV3 brick itself (where only 3 modes are available), the relection mode should give something like 2 or 3. So I need to investigate. Also, seems that some modes may be mixed. So need to investigate as well reason why.

In the mean time, can you confirm that you get the right EV3 sensor?

ClementG-63, 2019-12-16:
Yes it's this one

Ellerbach, 2019-12-16:
> Mode changedColor
> Brown
> 7% (not percentage)

In the color mode, it's normal that you get the color. The number is the color number (not a percentage) you get from the Lego EV3 brick itself. If you go to the port and select the mode with color, you'll get this.

ClementG-63, 2019-12-19:
I can’t get the color reflected percentage how you do it?
I plan to do line follower but looking at the project sources I don’t understand how you do to recover this percentage

Ellerbach, 2019-12-19:
> I can’t get the color reflected percentage how you do it?

It's by using the Reflection mode. But seems that there is an issue in the Read() function. I need to submit a fix. I don't have the sensor to test. But if you want to test, clone the project and change the Read function for the following:

```csharp
/// <summary>
        /// Read the intensity of the reflected or ambient light in percent. In color mode the color index is returned
        /// </summary>
        public int Read()
        {
            int val = 0;
            switch (_colorMode)
            {

                case ColorSensorMode.Color:                
                case ColorSensorMode.Ambient:
                    val = (int)ReadColor();
                    break;
                case ColorSensorMode.Reflection:
                default:
                    val = CalculateRawAverageAsPct();
                    break;
            }
            return val;
        }
```

If you don't want to clone the project and make any change, then you can still get the raw value thru the ```ReadTest()``` test function. The values return are space separated. The first one is the red index, the second one is the green index, the third one is the blue index and the fourth one is the background index which is the value you are looking for. 

so something like this will give you the right value:
```csharp
var str = sensor.ReadTest();
var percentage = Convert.ToInt32(str.Split(" ")[3]);
```

The variable ```percentage``` will contain the right value. Again, I haven't tested it as I don't have the hardware. I just had it for few hours to provide the first test.

Please let me know which way you'll go. You may want to check first what the ```ReadTest``` is returning and see if the value you are looking for is the proper one I pointed. If yes, that that should be a quick fix.

Ellerbach, 2021-07-01:
[Triage] I need to check if this fix has been merged or not, if not, then do it at some point.

krwq, 2023-01-19:
@Ellerbach ping :-)

PR #2020 (2023-01-22): Fixing EV3 color sensor
Fixes #882

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/2020)
closed 2023-01-22 by Ellerbach