dotnet/iot #883: About A4988 Stepper Motor Driver

gurelsoycaner, 2019-12-10:
Hi,
Just wondering when A4988 Stepper Motor Driver be available for use with .net core 3.0.

Note: Not with WindowsIOT but Ubuntu Raspberry pi

Thanks in advance.

joperezr, 2020-01-10:
@krwq Have you used one of these motors before? I wonder what do you m ean by "almost impossible"

krwq, 2020-01-10:
@joperezr I only tried to control stepper directly through H-bridge on Raspbian with different language and my experience was similar unless you rotated it very very slow. AR988 looks relatively simple to write a driver for. I think Uln2003 is conceptually similar to H-bridge so I expect experience to be similar.

I don't think I have any AR988 at the moment. @Ellerbach?

Ellerbach, 2020-01-12:
> I don't think I have any AR988 at the moment. @Ellerbach?

I'm travelling and not at home where I have a large base of components, so I need to check once back end of the week.

Uln2003 is a transistor array. So not really an H bridge. AR988 is made to create steper drivers up to sixteenth of step. To have something smooth to drive motors, better use a real driver.
And correct, should be quite straight forward to create a driver. See https://www.pololu.com/file/download/a4988_DMOS_microstepping_driver_with_translator.pdf?file_id=0J450

krwq, 2023-01-19:
[Triage] We're not tracking device binding requests. If you have anything to discuss around the binding please reopen

closed 2023-01-19 by krwq