dotnet/iot #885: Stepper motor

gurelsoycaner, 2019-12-11:
**Is your feature request related to a problem? Please describe.**

Describe what problem are you trying to solve or what's the expected outcome.

Can I possibly Pca9685 - I2C PWM Driver to work with STEPPER Motor like A4988 ?

If you'd like us to add a new device binding please add a full name and ideally links to specification / pictures of device you'd like to see here.

Please review [list of devices](https://github.com/dotnet/iot/blob/master/src/devices/README.md) to see if there is any similar device (perhaps belonging to the same family) which can be extended to also support your device.

**Describe the ideal solution**

A clear and concise description of what you want to happen.

nahueltaibo, 2019-12-11:
I think I can answer that. And it would be yes, using the [MotorHat device binding](https://github.com/dotnet/iot/tree/master/src/devices/MotorHat)... But we need to add the missing support for that.
The [MotorHat](https://www.adafruit.com/product/2348) can control 4 DC motors or **2 stepper motors** and it is based on the PCA9685.

I could add support for that if that is what you need.

krwq, 2019-12-11:
I haven't had a chance to use A4988 but if it's controlled by PWM then you should be able to use Pca9685 (not sure about MotorHat)

gurelsoycaner, 2019-12-12:
> 4

please, I really really need this support.

gurelsoycaner, 2019-12-12:
Please I need this support and will be very grateful if you can "add the missing support"

nahueltaibo, 2019-12-12:
Hey @krwq  how's the normal way of proceeding with this? should you assign this issue to me? or should I just work on my Fork and then make a pull request when I'm ready for it to be reviewed?

krwq, 2019-12-12:
@nahueltaibo as long as it's unassigned and it doesn't have a comment along the lines of "I'll work on this" it can be assumed that no one is working on it. I haven't seen anything explicit you want to pick this up so haven't assigned it but if the issue is related something you have worked on recently you and you have time to fix the issue just make sure everyone knows it's being worked on and you can go ahead and start working - if the change is touching System.Device.Gpio or something in Iot.Device.Bindings which is very popular then we should discuss the approach to avoid any changes we might not take (i.e. breaking changes or unnecessary refactoring will likely be a bit frowned upon while bug fixes are more than welcome).

nahueltaibo, 2019-12-12:
@krwq I see, OK, so: "I'll work on this" :)
Or at least I will be adding Stepper Motor capabilities to the MotorHat device binding I pushed some time ago, that is based on the Pca9685

gurelsoycaner, 2019-12-31:
> @krwq I see, OK, so: "I'll work on this" :)
> Or at least I will be adding Stepper Motor capabilities to the MotorHat device binding I pushed some time ago, that is based on the Pca9685

Hi,
Just wondering if any progress has been made

nahueltaibo, 2019-12-31:
Hi @gurelsoycaner, Not yet. I am in the middle of sitting for .net
certifications, so I'm studying in all my spare time.

Sorry. I will work on this once I have time again.

Cheers
*Nahuel Taibo*

gurelsoycaner, 2020-12-15:
Hi @nahueltaibo, hope you are well in the pandemic period? any luck for me with this issue, as you've promised to look into it?

raskolnikoov, 2020-12-25:
I also wonder when this could be released

krwq, 2021-01-11:
Given no response and no progress, assume this issue is free to be picked

Ellerbach, 2021-07-01:
[Triage] Up to grab! A nice one to start!

krwq, 2023-01-19:
[Triage] We're not tracking device binding requests. If you have anything to discuss around the binding please reopen

closed 2023-01-19 by krwq