dotnet/iot #916: Is any way to send command through UART port?

Parsakarami, 2019-12-24:
I want to send my command through the UART port to a module with asp.net core application which is run in Raspbian OS on Raspberry Pi 3.

When I use System.IO.Ports.SerialPort to open the serial port, I got UnathorizedAccessException. Is any way exist for send and receive command with UART port?

krwq, 2019-12-24:
1. Make sure to enable UART in raspi-config
2. change permissions of your serial port file so that you can run it as regular user (use udev to auto-change permissions) - as a quick test you can try running your app as sudo

joperezr, 2020-01-11:
@Parsakarami where you able to get it working after applying @krwq suggestions?

I'll go ahead and close this given this was a question and it was answered.

closed 2020-01-11 by joperezr

Parsakarami, 2020-01-12:
Thank you @krwq for your suggestions, I can change the permission of the Raspberry Pi UART port with the following command : 
       `sudo chmod 777 /dev/ttyS0`

To check permission also can use this command : 
       `ls -l /dev/ttyS0`

After set permission on the UART port, my problem has been solved.
Thanks.