dotnet/iot #918: Raspberry Pi GPIO pin numbering

SuperJMN, 2019-12-29:
Is there a rule to know with number corresponds to which pin in the GPIO port?

Thanks!

dpsenner, 2019-12-30:
I had to test the possibilities just a few hours ago and came to the following findings:

Looking at the [pin board](https://www.raspberrypi.org/documentation/usage/gpio/images/GPIO-Pinout-Diagram-2.png), the following applies:

* PinNumberingScheme.Board pin 11 is equivalent to PinNumberingScheme.Logical pin 17 (GPIO17)
* PinNumberingScheme.Board pin 13 is equivalent to PinNumberingScheme.Logical pin 27 (GPIO27)

You should by now see the pattern.

SuperJMN, 2019-12-30:
@dpsenner Thanks, but I think I'm not seeting something. Today I've confirmed that the pin number 17 corresponds to GPIO17. What am I missing?

pgrawehr, 2019-12-31:
Check this page: https://pinout.xyz/# The numbers noted as "BCM" are the ones you need if the scheme is set to logical (the default).

closed 2020-01-02 by SuperJMN