dotnet/iot #921: porting library with Apache 2.0 license

garciaolais, 2020-01-03:
Hi

I have a temperature 🌡  sensor [mlx90640 ](https://www.melexis.com/en/product/mlx90640/far-infrared-thermal-sensor-array) and I want to porting this code [melexis-mlx90640-library](https://github.com/melexis/mlx90640-library) but I am not sure if is possible porting code from Apache-2.0 to MIT. 

😕 ❓

krwq, 2020-01-03:
cc: @richlander

krwq, 2020-01-03:
@garciaolais any chance this sensor is in any way similar to https://github.com/dotnet/iot/tree/master/src/devices/Mlx90614 ? Possibly some of the code can be re-used. If that's the case it might be easier to compare the registers in the spec

joperezr, 2020-01-13:
We should use this issue so that once we have an answer for @garciaolais we should document it somewhere in our README. I believe the answer is that we can't take it, but let's wait for @richlander response.

krwq, 2020-01-14:
I think it actually might be allowed but might require 3rd party notice but I'd prefer @richlander to give official  answer

garciaolais, 2020-02-06:
Workout 🏋 
[image]

Windows Sample App with multiple temperature palettes 🎨 

[image]

----------------------------------------------------------------------------------------------------
I'm still working 🚧  on this device 📷 🌡  if someone finds 👀 this thread I can share the sample code. 😃 

PD. The sample is based on code with Apache license.

richlander, 2020-05-05:
You can port an Apache 2 library to MIT. You just need to an entry to https://github.com/dotnet/iot/blob/master/THIRD-PARTY-NOTICES.TXT.

closed 2020-05-05 by garciaolais