dotnet/iot #922: SoftwarePwmChannel should be in System.Device.Pwm

krwq, 2020-01-04:
Currently SoftwarePwmChannel exists in System.Device.Pwm.Drivers which is inconsistent with PwmChannel. It should be moved to System.Device.Pwm.

Please let me know what you think before I make any changes

cc: @joperezr @Ellerbach

shaggygi, 2020-01-05:
Agreed.  I’m assuming this will eventually be moved to the core library (similar to what you previously mentioned about moving the software SPI).  If that is not the intention, it might be worth to go ahead and also rename SoftPwm folder to SoftwarePwm.

krwq, 2020-01-05:
@shaggygi yes, I hope it will be moved eventually although I think it should pass all our tests first which currently it doesn't (Windows is running against it as fallback).

joperezr, 2020-01-17:
I agree with @shaggygi and would only do this once we can a) rename this to SoftwarePwm and b) move it down to System.Device.Gpio. That is the right place for it anyway and I think doing these 3 steps separately makes no sense, so lets just do all three together.

pgrawehr, 2022-12-10:
The name change is no problem at all, but do we really want to move it to System.Device.Gpio? We should keep that library as small as possible IMHO.

krwq, 2022-12-15:
I'm fine with closing this

closed 2022-12-15 by krwq