dotnet/iot #924: PCA9685 unreliable/broken

IVData, 2020-01-05:
**Describe the bug**

It seems like since #692 the PCA9685 no longer works correctly. Motors either don't spin at all, or they spin very slowly.

**Steps to reproduce**

```csharp
        private static Pca9685 device;
        private static PwmChannel[] channels = new PwmChannel[16];
        public PCA9685_MotorController()
        {
            var busId = 1;  // /dev/i2c-1
            var deviceAddress = 0x40;
            var settings = new I2cConnectionSettings(busId, deviceAddress);
            var d = I2cDevice.Create(settings);
            device = new Pca9685(d);
            Console.WriteLine($"PWM Frequency: {device.PwmFrequency}Hz");
            Console.WriteLine($"Dutycycle: {device.GetDutyCycle(1)}");

            for (int i = 0; i < 16; i++)
            {
                channels[i] = device.CreatePwmChannel(i);
            }
        }

        public async Task RunMotorForMsAsync(int channel, int ms, double speed = 1)
        {
            Console.WriteLine($"Running channel {channel} for {ms}ms with dutycycle {speed}");
            Console.WriteLine($"Dutycycle: {device.GetDutyCycle(channel)}");
            channels[channel].DutyCycle = speed;
            Console.WriteLine($"Dutycycle after pwmChannel dutycycle set to {speed}: {device.GetDutyCycle(channel)}");
            channels[channel].Start();
            Console.WriteLine($"Dutycycle after motor start: {device.GetDutyCycle(channel)}");
            await Task.Delay(ms);
            channels[channel].Stop();
            Console.WriteLine($"Dutycycle after motor stop: {device.GetDutyCycle(channel)}");
        }
```
Results in this output:

```
PWM Frequency: 1017.2526041666666Hz
Dutycycle: 1
Running channel 8 for 520ms with dutycycle 1
Dutycycle: 0
Dutycycle after pwmChannel dutycycle set to 1: 1
Dutycycle after motor start: 1
Dutycycle after motor stop: 0
Running channel 8 for 1300ms with dutycycle 1
Dutycycle: 0
Dutycycle after pwmChannel dutycycle set to 1: 0
Dutycycle after motor start: 1
Dutycycle after motor stop: 0
Running channel 9 for 520ms with dutycycle 1
Dutycycle: 0
Dutycycle after pwmChannel dutycycle set to 1: 1
Dutycycle after motor start: 1
Dutycycle after motor stop: 0
```

The duty cycle parameters are sometime set and sometimes not. When the motors do work (I have't worked out what gets me *some* movement yet), they run slow.
**Expected behavior**

Motors should perform the same as in my old preview version (sorry not sure which version I had).

**Actual behavior**

Motors usually don't react at all.

**Versions used**

Add following information:

```
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100
 Commit:    04339c3a26

Runtime Environment:
 OS Name:     raspbian
 OS Version:  9
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/dotnet/sdk/3.0.100/

Host (useful for support):
  Version: 3.0.0
  Commit:  7d57652f33

.NET Core SDKs installed:
  3.0.100 [/home/pi/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.0.0 [/home/pi/dotnet/shared/Microsoft.NETCore.App]
```
iot.Device.Bindings 1.0.0
System.Device.Gpio 1.0.0

closed 2020-01-05 by IVData

krwq, 2020-01-05:
@IVData can you share what problem have you hit so that we can potentially improve docs/sample?

IVData, 2020-01-05:
Of course, as soon as I posted this I started suspecting a hardware issue, and found out the grounds between my PCA9685 and motor drivers were drifting apart, causing the weak drive and sometimes no drive at all.

I've fixed the issue and as far as I can tell this library is working perfectly. I just happened to replace the power supplies and upgrade packages at the same time. So apologies for the incorrect bug report.

krwq, 2020-01-05:
No problem, it's hardware - this happens a lot to everyone :smile: