dotnet/iot #929: SoftwarePwm should pick precision timer automatically

krwq, 2020-01-06:
By default SoftwarePwm does not use precision timer (precise while loop) which is causing that you can't exactly get reliably frequency - that is fine for servo motors but not exactly for any other use. Under oscilloscope I'm getting around 50Hz (shifts between 47.2Hz - 50Hz with most time spent exactly on 50Hz) by default even though it should be 400Hz.

We should consider removing this argument and automatically pick the right method based on the passed in frequency since for anything other than servo motor this is likely not satisfactory.

cc: @Ellerbach

krwq, 2020-01-06:
I've just checked how does the duty cycle perform for 50Hz and except for 0.0/0.5/1.0 all values seem to be inconsistent, i.e. for 0.7 I'm getting more or less 1/4 pulses with somewhat random width - my oscilloscope shows duty cycle as anything between 58-72% with average around 65%... I'm wondering if we should just not always use high precision timer

pgrawehr, 2020-01-06:
I have a weird feeling about the usage of 
```
Task.Delay(TimeSpan.FromMilliseconds(_currentPulseWidth)).Wait();
```
in the code. Seems needlessly complicated just for waiting. Can you try whether these changes I prepared sometime ago (but didn't really complete, because I considered it not so urgent) improve the behavior in the default mode? https://github.com/pgrawehr/iot/tree/SoftwarePwmImprovement

In general, I'm not sure we should change the default to high-precision mode, mostly because that eats a lot of CPU (basically one CPU core per Software PWM Channel instance). If you just want to drive an RGB LED using 3 Software PWM Channels, that's going to eat almost all the resources you have on a Pi. As option, we might change the implementation in such a way that it only uses _one_ thread for all active instances.

krwq, 2020-01-07:
@pgrawehr I've just spoke with someone from JIT team and neither AutoResetEvent nor Task.Delay nor Thread.Sleep guarantee anything better than few ms precision. I'd recommend to use DelayHelpers from our code instead for both methods and perhaps allow thread yields when no precision is required. This way we at least have some way of swapping underlying implementation globally

pgrawehr, 2020-01-08:
I guess you are right. At 50Hz, one cycle is 20ms, so at duty 0.5, that's one event every 10ms, which is already at the minimum thread switch time for some CPU/OS combinations.

Ellerbach, 2020-01-09:
> I guess you are right. At 50Hz, one cycle is 20ms, so at duty 0.5, that's one event every 10ms, which is already at the minimum thread switch time for some CPU/OS combinations.

Yes, the non precise one is really to be used with led or some motors equivalent where it doesn't really matter if it's extremely precise. The advantage is that the thread and processor usage is very low in this mode. 
The precise mode has to be used for servo motors for example, as @krwq mentioned in one of the thread, it can get much higher, couple of kHz. The initial code wasn't really designed to get to high level frequencies, it can for sure be optimized. I mainly used it with servo motors. And inconvinient is that it is using a lot of processor. That's the price to pay.

pgrawehr, 2020-01-09:
What is the maximum frequency that is used to drive motors?

I think we should at least consider creating one high-precision thread for all SoftwarePwm Channels. With a bit of intelligent delay compuation, we could at least limit the high CPU load to one core.

krwq, 2020-01-09:
@pgrawehr for DC motor I never needed more than 50Hz and SoftPWM with non-accurate timer was working fine for me. For servos I wouldn't expect non-precise to work correctly. IMO we should start with making current SoftPWM implementation better and make sure it passes the tests in my PR which upgrades the CI setup. Once we are confident this works correctly and have CI testing that then we should optimize for multiple SoftPWM channels while making sure main scenario still works well.

pgrawehr, 2020-01-09:
I agree on that - that was just an idea for improvement to limit the effect of the most obvious downside of the precise implementation.

krwq, 2020-01-09:
@pgrawehr eventually we can perhaps also join that with some of the event logic used for waiting but not sure how much we can squeeze into a single thread without losing precision too much

pgrawehr, 2020-01-09:
Yea, also had that idea. I'll try to do some tests on CPU load with interrupt driven approach. I have verified that interrupts at 1khz can be handled with reasonable precision, but I didn't check the CPU load for doing that.

pgrawehr, 2020-01-09:
[Here](http://www.netzmafia.de/skripten/hardware/RasPi/timer/index.html) is a very extensive documentation about timers and achievable delays in linux and on the Raspi in particular. Not sure it helps us anything, but that guy there seems to know exactly how the linux kernel handles timers. The text is in german, tough.

closed 2022-12-15 by krwq