dotnet/iot #937: CI Status badge linking to badge image

eiriktsarpalis, 2020-01-13:
**Describe the bug**

The [CI status badge](https://github.com/dotnet/iot#official-build-status) is linking to the badge itself as opposed to some pipeline.

joperezr, 2020-01-13:
I believe you are correct, thanks for pointing it out @eiriktsarpalis I'll fix this right away.

joperezr, 2020-01-13:
Ohh I see what the problem is, the problem is that I linked it to the actual official build status which is internal so only signed in users can see that. I'll switch it to the open rolling build instead.

closed 2020-01-13 by joperezr

krwq, 2020-01-13:
@joperezr now it shows "never built"

joperezr, 2020-01-13:
Yup, that's expected, continuous integration was disabled on the public build, I enabled that but in order for it to run it needed at least one new commit, which was the one for my readme. So it should be either green/red on a few mins :)

PR #938 (2020-01-13): Fixing the Pipeline badge
Fixes #937 
cc: @eiriktsarpalis
commit d07e5ecc (2020-01-13): Fixing the Pipeline badge (#938)

Fixes #937