dotnet/iot #941: Flaky tests in CI

krwq, 2020-01-14:
I've just run (Linux Release) CI tests 50 times in a row and out of 50 runs 6 failed with following failures:

5x:

```
    System.Device.Gpio.Tests.RaspberryPiDriverTests.InputPullResistorsWork [FAIL]
      Assert.Equal() Failure
      Expected: Low
      Actual:   High
      Stack Trace:
        /_/src/System.Device.Gpio.Tests/RaspberryPiDriverTests.Linux.cs(34,0): at System.Device.Gpio.Tests.RaspberryPiDriverTests.InputPullResistorsWork()
```

1x:

```
    System.Device.Gpio.Tests.SysFsDriverTests.AddCallbackFallingEdgeNotDetectedTest [FAIL]
      Assert.False() Failure
      Expected: False
      Actual:   True
      Stack Trace:
        /_/src/System.Device.Gpio.Tests/GpioControllerTestBase.cs(176,0): at System.Device.Gpio.Tests.GpioControllerTestBase.AddCallbackFallingEdgeNotDetectedTest()
```

cc: @pgrawehr @joperezr

We should try to understand the failures and if this is by design then retry or remove the test

pgrawehr, 2020-01-14:
Eventing will be way better with #914, but I couldn't get it 100% reliable as well.

krwq, 2020-01-14:
I'm generally fine with retries on SysFS eventing implementation since we understand there is a limitation but for the PullUp/PullDown I'd expect this to just work and I'm uncertain why the test is failing and I'm a bit concerned with hiding the issue. As a temporary solution retrying seems fine to me but we should create an issue for investigation

krwq, 2020-01-14:
I'm wondering if perhaps this delay isn't too small:
https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/Drivers/RaspberryPi3Driver.Linux.cs#L277

And validate if spec
https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf Section "GPIO Pull-up/down Clock Registers (GPPUDCLKn)" have some extra info which perhaps we missed..

krwq, 2020-01-14:
Perhaps we should also not store the temporary value in the local since that is opening time window where something else might be fighting with the value of the register

pgrawehr, 2020-01-15:
@joperezr has opened #943 to track that test explicitly. We should discuss specifics there.

krwq, 2020-01-17:
After merging pull up/down PR here are the stats.

3/100 failed with all 3 failures being:

```
    System.Device.Gpio.Tests.SysFsDriverTests.AddCallbackFallingEdgeNotDetectedTest [FAIL]
      Assert.False() Failure
      Expected: False
      Actual:   True
      Stack Trace:
        /_/src/System.Device.Gpio.Tests/GpioControllerTestBase.cs(176,0): at System.Device.Gpio.Tests.GpioControllerTestBase.AddCallbackFallingEdgeNotDetectedTest()
```

krwq, 2020-04-21:
Since we haven't had any progress on the remaining failure I'm voting on disabling the test on scenarios which don't work correctly and creating an issue on it (if it doesn't exist already). cc: @joperezr

pgrawehr, 2020-04-22:
I agree for now. I was already trying to improve this test (when I was doing the performance analysis) but could only improve it, but apparently not fully fix it. It seems it has to do with some pending interrupts at the time the event callback is set up, but the picture is not so clear.

closed 2023-03-16 by krwq