dotnet/iot #943: Test InputPullResistorsWork will fail intermitently

joperezr, 2020-01-14:
During some sample test runs, @krwq discovered that this test will fail 12% of the time which is why as a mitigation we opted to do #942 which adds a few retries in case the test fails. This workitem is going to track the work of investigating what is wrong with that test and to provide a fix for it in order to make it more stable, or to fix the product if that is appropriate. 

cc @krwq @pgrawehr

pgrawehr, 2020-01-15:
Question: Do all the CI tests run on Pi 3's? Because, as I mentioned, the problem sometimes happens on my board too, and that is a Pi4, which has entirely different logic for setting up the pull resistors. This makes it kind of unlikely that it's only a timing issue in the code for changing the settings (which is pretty complicated for the Pi3).

krwq, 2020-01-15:
@pgrawehr, yes, all tests run on Pi3s at the moment.

Moving my notes from the other issue (things we should try here):
- not store the value of the register in the local since that is opening time window where something else might be fighting with the value of the register, we should operate on it directly
- validate if spec
https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf Section "GPIO Pull-up/down Clock Registers (GPPUDCLKn)" have some extra info which perhaps we missed
- check if perhaps this delay isn't too small:
https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/Drivers/RaspberryPi3Driver.Linux.cs#L277 - does SpinWait actually wait in cycles?

pgrawehr, 2020-01-15:
Yes, SpinWait is clock-dependent: https://docs.microsoft.com/en-us/dotnet/api/system.threading.thread.spinwait?view=netframework-4.8 

Maybe we should calculate the duration of a cycle for SpinWait on app startup?

PR #945 (2020-01-16): Fix Pull Up/Down intermittently not working
Fixes #943
Contributes to #941

Did couple of changes before all failures were gone. Out of 50 CI runs 1 failed and issue was unrelated (the other intermittent failure from #941).

Tested only on Release build.

When testing number of iterations in the test was increased to 10k

Changes made:
- added zeroing clock (this fixed most of the problems)
- operating directly on pointer
- despite spec order zeroing clock happens first since that seem to have causing intermittent issues as well
- SpinWait was replaced with regular loop and optimizations in th [... cut]
closed 2020-01-16 by krwq