dotnet/iot #948: Bluetooth Low Energy support?

Ellerbach, 2020-01-16:
Some devices are connected using Bluetooth Low Energy (BLE). This is supported in Xamarin for example on the diverse platforms. Is there any plan to have similar support for .NET Core?

krwq, 2020-01-16:
We haven't considered this during planning. Do we have any sense on how much work that is?

shaggygi, 2020-01-16:
Seems like it would follow similar pattern as System.Device.Gpio (and protocols - SPI, I2C, etc.) like [Windows.Devices.Bluetooth](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth)... System.Device.Bluetooth?  Should this be a separate package as it most likely wouldn't need dependencies on current GPIO stuff?

krwq, 2020-01-17:
I'd start with putting this in Iot.Device.Bindings bag for a starter and once it's stable enough either put it in System.Device.Gpio or new package

Ellerbach, 2020-01-17:
@krwq, I don't know exactly how much work it will require. Looking at some part of Xamarin and couple of additional packages, it's quite some work. BLE seems to have quite some specificities with contracts and services, the mechanism needs to be implemented quite low level on my understanding and then there will be the possibility to add capabilities for specific sensors like buttons, like temperature, some integrated portable devices, etc. 
This clearly needs more research. And yes, probably for a new phase. But I see more and more sensors integrating this capability to be wireless and low consumption.

hansmbakker, 2021-01-17:
For the Linux side, there is a .net library in development at https://github.com/hashtagchris/DotNet-BlueZ.

There is a lot of abstraction work in `Windows.Devices.Bluetooth` which could be taken as an example, but the platform-specific implementation would then still need to be made.

hansmbakker, 2021-01-18:
There is also work in https://github.com/nexussays/BLE.net for other platforms

danielmeza, 2021-03-01:
@hansmbakker DotNet-BlueZ is a good library for build BLE Clients apps. I have already implemented an server app using  [dbus](https://github.com/tmds/Tmds.DBus) to implement BlueZ also. But before to use bluez as a solution on Linux systems, we need to include dbus support on .net core, currently we had [Tmds.DBus](https://github.com/tmds/Tmds.DBus) implementation on .net core but it not below to any foundation and will be great if we include that firs on .net core as a dbus implementation. If not, we need to make another abstraction layer.

raffaeler, 2023-09-05:
I had a brief conversation with the fellow MVP Peter Foot @inthehand about bluetooth support.
He is working on BT (classic and low-energy) support in .NET and he already implemented the fundamental parts (see https://github.com/inthehand/32feet).

Once it's completed, it would be interesting creating a layer in our library to abstract devices behind BT.

Ellerbach, 2023-09-05:
Fantastic news @inthehand and @raffaeler ! What would be even better is to have the API aligned with the .NET nanoFramework implementation as we already have quite some people using it: https://github.com/nanoframework/nanoFramework.Device.Bluetooth
We already have SPI, GPIO, I2C, Serial aligned ;-) And this time, it's first on nano. So IoT should align with it.

raffaeler, 2024-06-05:
@inthehand Peter, could you please provide some news about the work you were doing in this space?
Thanks!