dotnet/iot #963: Update IoT.Device.Bindings

richlander, 2020-01-29:
It looks like the package is now out of date.

This line breaks with the current NuGet package: https://github.com/dotnet/iot/blob/samples/src/devices/Bmxx80/samples/Bme280.sample.cs#L9

We need to define a release approach for this assembly, including how to report breaking changes.

joperezr, 2020-01-29:
Yes, @krwq and I have talked about this, and we have an easy way to check all breaking changes that have happened on the iot.device.bindings package, so we will communicate all of these for our next release.

elead1, 2020-02-03:
Any timeline on the next release? The current master branch indicates [support for 4-bit I2C devices](https://github.com/dotnet/iot/blob/2ceabde76bfc094678432a2faa67beaf7c513d78/src/devices/CharacterLcd/LcdInterface.cs#L145), but the release 1.0.0 only supports 8-bit. I worked around this issue by installing a pre-release version 1.1.0-prerelease.19577.1, but it seems the pre-release may have been removed from the index?

joperezr, 2020-02-03:
> Any timeline on the next release?

The full release of vNext is not fully decided yet, but it is very possible that it will just align with .NET 5 schedule (meaning we will probably ship a stable version of the packages into NuGet around November). That said, we plan on starting to ship previews of vNext into NuGet starting this month, and they should go in in a monthly cadence.

> it seems the pre-release may have been removed from the index?

It hasn't just make sure that you are pointing into our nightly build feeds: `https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json`

joperezr, 2020-03-03:
We have just released our first preview for vNext https://www.nuget.org/packages/System.Device.Gpio/1.1.0-prerelease.20153.1

I'll keep this issue open so that I update the samples to consume these new package versions.

pgrawehr, 2020-03-05:
@joperezr Do we have a specific ticket to track what is left for vNext or to discuss what should be in there? The content of the milestone doesn't seem to be complete.

joperezr, 2020-03-05:
We do, here is our [Roadmap](https://github.com/dotnet/iot/blob/master/Documentation/roadmap.md). You can also check the issues we currently have marked as vNext with this query: https://github.com/dotnet/iot/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc+milestone%3AvNext. If you feel like there is something else we should include in vNext, feel free to open an issue to start the discussion, and we can triage it and set the milestone appropriate.

closed 2022-12-08 by krwq