dotnet/iot #971: UnhandledException in WasEventDetected method

shuebener, 2020-02-04:
On my test system the method _WasEventDetected_ from the _SysFsDriver.Linux.cs_ throws an Exception because the _Interop.epoll_wait_ method fails. While I haven't found the reason yet, how am I supposed to catch this exception? Since this code is executed in a new internal Thread, it will jump to my unhandled exception handler and kills my application.

joperezr, 2020-02-04:
Thanks for logging this @shuebener, you are correct that we are running that code in a separate thread and we are not catching exceptions raised there so we should probably fixed that. That said, I'm not sure which exception might be thrown from epoll_wait, so if you have a repro or if you could attach to the process, that would be great so that we can better understand what is going on.

shuebener, 2020-02-05:
https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/Drivers/SysFsDriver.Linux.cs#L409
```
// poll every 1 millisecond
int waitResult = Interop.epoll_wait(pollFileDescriptor, out epoll_event events, 1, _pollingTimeoutInMilliseconds);
if (waitResult == -1)
{
	throw new IOException("Error while trying to initialize pin interrupts.");
}
```

It happens for me only a few times, when epoll_wait returns -1. 
The errno then is EINTR (4) -> "The call was interrupted by a signal handler before the timeout expired or any requested event took place." 
I noticed, there is a SetLastError missing for the epoll_wait dll import. I it should be added there too, to have the error code available.

To me it seems to happen occasionally when there is a higher CPU load on my board. So it's just like a simple timeout. So for now, I only removed the generation of the exception there and it's working fine with this.

joperezr, 2020-02-07:
I see yeah that makes sense, we definitely have to fix that one. If you happen to get a repro that would be great so that we can check what is errorno set to, since according to this http://man7.org/linux/man-pages/man2/epoll_wait.2.html it should set errno correctly when -1 is returned.

PR #1801 (2022-02-24): retry when epoll_wait fails with EINTR
Fixes #971

Implement retry handling for `epoll_wait` in `SysFsDriver.WasEventDetected` similar to `LibGpiodDriverEventHandler` implementation.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1801)
closed 2022-02-24 by joperezr