dotnet/iot #974: PinValueChangedEventArgs uses wrong PinNumberingScheme

dontspeak, 2020-02-06:
Im initiating the gpiocontroller with **board** numbering scheme, but when I receive change events, the `PinValueChangedEventArgs  `does provides the **logical** pin:

```
GpioController controller = new GpioController(PinNumberingScheme.Board);
controller.OpenPin(16, PinMode.InputPullUp); // Set "16"
controller.RegisterCallbackForPinValueChangedEvent(16, PinEventTypes.Falling, MyEvent);

public void MyEvent(object sender, PinValueChangedEventArgs args)
{
Log(args.PinNumber); // Gives me "23"
}
```

The `PinNumber `should depend on the `PinNumberingScheme`? I.E. in that case, it should return the according board pin number (which is 16) and not the logical one (which is 23).

pgrawehr, 2020-02-06:
I guess you are right. From what I've seen so far, the board numbering scheme is only marginally supported.

joperezr, 2020-02-07:
Thanks for logging this @dontspeak, that was a great find! I believe you are right, we usually just handle the numbering scheme on the controller, but all the actual drivers use logical numbering so I don't doubt that this is the case today.

Ellerbach, 2021-07-01:
[Triage] Closing this one as addressed on the various PR

closed 2021-07-01 by Ellerbach