dotnet/iot #984: DHT22 Wrong temperature readings.

BramEsendam, 2020-02-19:
The temperature output dropped to to 1C wen the real temperature was 26C.

**Steps to reproduce**

```csharp
using var dht = new Dht22(26);
while (true)
 {
         // Try to read the temperature.
         var temp = dht.Temperature;
         if (!dht.IsLastReadSuccessful) continue;

          // Try to read the humidity.
          var humidity = dht.Humidity;
          if (!dht.IsLastReadSuccessful) continue;

          Console.WriteLine($"Temperature: {temp.Celsius:0.0} °C, 'Humidity: {humidity:0.0} %");

          break;
}
```

**Expected behavior**

Output: Temperature: 26.0 °C, 'Humidity: 36.1 %

**Actual behavior**

Output: Temperature: 1.0 °C, 'Humidity: 36.1 %

**Versions used**

Add following information:

- Dotnet on the build machine: `Version: 3.1.1 Commit:  a1388f194c`
- Dotnet on the raspberry pi 3B: `Version: 3.1.1 Commit:  a1388f194c`
- Version of `System.Device.Gpio` package: `1.0.0`
- Version of `Iot.Device.Bindings` package: `1.0.0`

BramEsendam, 2020-02-19:
[image]

I ran it with a small python script as well.
And it resulted in the correct values.

BramEsendam, 2020-02-19:
I was running a Dht11 sensor before this, and i never had any problems with it.
Except of the 2c inaccuracy :P

joperezr, 2020-02-19:
> I was running a Dht11 sensor before this, and i never had any problems with it.

So when you say ya were running it before, do you mean with python or with .net core? If you were doing it with .net core and it was working correctly, did you recompile or updated the nuget package to get a newer version?

cc: @Ellerbach @ZhangGaoxing as FYI.

BramEsendam, 2020-02-20:
I was running the same code. Except for using the dht22 class i was using the dht11 one.

fomenkodmitry, 2020-03-07:
Sustained, similar bug.
You can see the code here: https://github.com/fomenkodmitry/AspNetCore-raspberrypi4-dht22
Platform: Rasberry pi 4 model b, net core 3.1

```bash
pi@raspberrypi:~/AspNetCore-raspberrypi4-dht22/NetCoreRaspb $ python3 ../../humidity.py 
Temp=26.3*C  Humidity=33.4%
Temp=26.3*C  Humidity=33.9%
Temp=26.3*C  Humidity=33.9%
```
```bash
dotnet run
Hello DHT!
Temperature: NaN°C, Humidity: NaN%
Temperature: NaN°C, Humidity: NaN%
Temperature: NaN°C, Humidity: NaN%
Temperature: 1.7°C, Humidity: 33.2%
Temperature: 1.7°C, Humidity: 33.3%
````

Ellerbach, 2020-03-12:
@fomenkodmitry and  @BramEsendam, you are right, there was a mistake. I found some time to dig into the code and found the problem.
Line 32 in the Dht222.cs file should be:
```csharp
var temp = ((readBuff[2] & 0x7F) * 256 + readBuff[3]) * 0.1;
```
@joperezr, will make a PR asap.

joperezr, 2020-03-12:
I've merged that PR so hopefully the results will be fine now. I'll go ahead and close this issue assuming it does, but please feel free to reopen if that is not the case.

closed 2020-03-12 by joperezr