dotnet/iot #987: Consider GDI+ for Ssd13xx

A-J-Bauer, 2020-02-21:
Consider the usage of GDI+ functions for the Ssd13xx displays.
Please have a look at [OledI2c128x64SH1106](https://github.com/A-J-Bauer/OledI2c128x64SH1106)

krwq, 2020-02-21:
System.Drawing is using GDI+ under the covers, perhaps we can plug it in there if it's not yet there

joperezr, 2020-02-21:
My idea was that because we were already using System.Drawing we were already using GDI+ as @krwq mentions. Is there something specific in mind you had where we could take different advantage of GDI+?

A-J-Bauer, 2020-02-22:
I was thinking of an approach like the one used in WinfForms where OnPaint is provided which exposes a System.Drawing.Graphics object. OnPaint that can be triggered with Refresh().

Exposing a System.Drawing.Bitmap and an Update method/function (without events) would suffice though.

All drawing, shapes, lines images, text, scrolling etc. could then be done by using the graphics object/bitmap.

Something like this: [working sample code for a sh1106 display](https://github.com/A-J-Bauer/OledI2c128x64SH1106/blob/master/OledI2c128x64SH1106.cs) (Update function at line 162)

Frankenslag, 2020-02-22:
Just FYI the SSD1351 already supports the use of System.Drawing.Bitmap if that is any help.

A-J-Bauer, 2020-02-22:
Ok didn't know that. I probably didn't look at the existing code close enough then. Thanks.

closed 2020-02-22 by A-J-Bauer