dotnet/iot #988: Unable to resolve dependency 'System.Memory' when trying to use nightly

alvipeo, 2020-02-21:
I have an existing project which "works on my machine" (compiles on my machine and works on Rasp Pi). And I wanted to build it on another machine. On this another machine I tried to install nightly pre-release builds of the packages but it fails with the following error:

>Unable to resolve dependency 'System.Memory'. Source(s) used: 'nuget.org', 'Microsoft Visual Studio Offline Packages'.

This is the first time I get this error. Does anybody know how to fix this? Is this just a temporary bug?
Thanks.

krwq, 2020-02-21:
can you please share `dotnet --info` from the machine where you see issues and the exact steps you did to get this error? (if you can repro with new project please share template used for `dotnet new ...` if you can't please share .csproj file content)

alvipeo, 2020-02-21:
When I cloned my project on this machine and run `dotnet restore` I get:

>Core.csproj : error NU1102: Unable to find package System.Device.Gpio with version (>= 1.1.0-prerelease.20052.2)

So I try to get the package from nuget:

```
nuget install System.Device.Gpio -PreRelease -Source https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json
Feeds used:
  https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json

Installing package 'System.Device.Gpio' to 'D:\...Core'.
  GET https://pkgs.dev.azure.com/dnceng/9ee6d478-d288-47f7-aacc-f6e6d082ae6d/_packaging/7d9f5c21-0d79-403f-bfe3-9a4506529760/nuget/v3/registrations2-semver2/system.device.gpio/index.json
  OK https://pkgs.dev.azure.com/dnceng/9ee6d478-d288-47f7-aacc-f6e6d082ae6d/_packaging/7d9f5c21-0d79-403f-bfe3-9a4506529760/nuget/v3/registrations2-semver2/system.device.gpio/index.json 334ms

Attempting to gather dependency information for package 'System.Device.Gpio.1.1.0-prerelease.20119.1' with respect to project 'D:\...Core', targeting 'Any,Version=v0.0'
Gathering dependency information took 4,45 sec
Attempting to resolve dependencies for package 'System.Device.Gpio.1.1.0-prerelease.20119.1' with DependencyBehavior 'Lowest'
__Unable to resolve dependency 'System.Memory'. Source(s) used: 'nuget.org', 'Microsoft Visual Studio Offline Packages'.__
```

dotnet --info (**what?? I haven't paid attention to this...**):

>.NET Core SDK (reflecting any global.json):
 Version:   3.1.101
 Commit:    b377529961

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.18363
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\3.1.101\

Host (useful for support):
  Version: 3.1.1
  Commit:  a1388f194c

.NET Core SDKs installed:
  2.1.1-preview-007165 [C:\Program Files\dotnet\sdk]
  2.1.1 [C:\Program Files\dotnet\sdk]
  2.1.2 [C:\Program Files\dotnet\sdk]
  2.1.4 [C:\Program Files\dotnet\sdk]
  2.1.100-preview-007326 [C:\Program Files\dotnet\sdk]
  2.1.100-preview-007354 [C:\Program Files\dotnet\sdk]
  2.1.100-preview-007363 [C:\Program Files\dotnet\sdk]
  2.1.100 [C:\Program Files\dotnet\sdk]
  2.1.200-preview-007474 [C:\Program Files\dotnet\sdk]
  2.1.200-preview-007517 [C:\Program Files\dotnet\sdk]
  2.1.200-preview-007576 [C:\Program Files\dotnet\sdk]
  2.1.200-preview-007589 [C:\Program Files\dotnet\sdk]
  2.1.200-preview-007597 [C:\Program Files\dotnet\sdk]
  2.1.200 [C:\Program Files\dotnet\sdk]
  2.1.201 [C:\Program Files\dotnet\sdk]
  2.1.202 [C:\Program Files\dotnet\sdk]
  2.1.300-rc1-008673 [C:\Program Files\dotnet\sdk]
  2.1.300 [C:\Program Files\dotnet\sdk]
  2.1.301 [C:\Program Files\dotnet\sdk]
  2.1.400-preview-009063 [C:\Program Files\dotnet\sdk]
  2.1.400-preview-009171 [C:\Program Files\dotnet\sdk]
  2.1.400 [C:\Program Files\dotnet\sdk]
  2.1.401 [C:\Program Files\dotnet\sdk]
  2.1.402 [C:\Program Files\dotnet\sdk]
  2.1.500-preview-009297 [C:\Program Files\dotnet\sdk]
  2.1.500-preview-009335 [C:\Program Files\dotnet\sdk]
  2.1.500-preview-009404 [C:\Program Files\dotnet\sdk]
  2.1.500 [C:\Program Files\dotnet\sdk]
  2.1.502 [C:\Program Files\dotnet\sdk]
  2.1.503 [C:\Program Files\dotnet\sdk]
  2.1.504 [C:\Program Files\dotnet\sdk]
[... 95 lines cut ...]
  Microsoft.NETCore.App 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.9 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.11 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.14 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.15 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.3 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.4 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.5 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.0 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.1 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download

joperezr, 2020-02-21:
mm I'm not sure about this one, to me it just looks like a networking issue. Just to make sure you are using the right feeds, can you please run the following command form the folder containing your .csproj:
`dotnet new nugetconfig`. This should create a new NuGet.config file and place it on that directory which will ensure that your package sources are looking into the right nuget.org source.

alvipeo, 2020-02-22:
here's nuget.config that was created:

```
<?xml version="1.0" encoding="utf-8"?>
<configuration>
  <packageSources>
    <!--To inherit the global NuGet package sources remove the <clear/> line below -->
    <clear />
    <add key="nuget" value="https://api.nuget.org/v3/index.json" />
  </packageSources>
</configuration>
```

alvipeo, 2020-02-22:
Tried it again after nuget.config was created - same thing, same error.

joperezr, 2020-02-24:
You should add the iot nightly feed to the package sources if you want to download nightly builds:
```xml
    <add key="dotnet5" value="https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json" />
  </packageSources>
```

And about nuget.org not working, can you try adding this after packageSources on the NuGet.config?

```xml
  ...
  </packageSources>
  <disabledPackageSources>
    <clear />
  </disabledPackageSources>
```

The only thing that comes to mind is that you might have unchecked nuget.org source from VS, and that is a setting that takes effect machine-wide so even with a local nuget.config adding the source, it won't actually use it.  By adding the above code to your nuget.config, then it will clear those machine-wide settings so it will force to use nuget.org again.

If that doesn't fix your issue, I would suggest logging an issue on the NuGet repo as

alvipeo, 2020-02-26:
Adding this:
```
    <add key="dotnet5" value="https://pkgs.dev.azure.com/dnceng/public/_packaging/dotnet5/nuget/v3/index.json" />
```
helped! Thank you.

closed 2020-02-26 by joperezr