dotnet/iot #996: Build Error with System.Devices.Spi & Functions associated

ClementG-63, 2020-03-01:
When I build my code with sources (not Nugget) I have the following error: "CS0234: TypeName or Namespace "Devices" doesn't exist in the namespace 'System')

Then I can't call any function or type like          
       `var settings = new SpiConnectionSettings(busId, ChipSelectLine);`

**Steps to reproduce**

```csharp
using System.**Devices**.Spi;
private static **SpiDevice** _brickPiSPI = null;
```
(** is where code is red-underlined

**Versions used**

Add the following information:

- `dotnet --info` on the machine being used to build
-> .netcore3.1

krwq, 2020-03-02:
@ClementG-63 any chance you could provide exact steps on how you setup a project and how you build it?

joperezr, 2020-03-02:
The namespace is `Device`, not `Devices` which might be why you are seeing this problem.

joperezr, 2020-03-04:
Given that I believe the problem was just that the namespace was wrong here, I'll go ahead and close the issue, but please @ClementG-63 feel free to reopen if that is not the case.

closed 2020-03-04 by joperezr