dotnet/iot #998: Support for Rasp PI W Zero - possible?

alvipeo, 2020-03-03:
Hi guys!

I've used .Net Core 3.1 on Raspberry Pi 3 and it works like a charm! So thank you very much for that!

But I wonder if it's possible to run .Net Core 3.1 apps on Raspberry Pi W Zero? This would be really REALLY powerful and very cost effective for simple but important apps. I'm sure so many devs would be greatfull.

Thanks!

joperezr, 2020-03-03:
Thanks a lot for the feedback and we are glad that the project has worked for you!

Unfortunately, .NET Core doesn't run on Raspberry Pi Zero, but the good news is that other .NET runtimes like Mono does, and because our library targets .NET Standard 2.0, that means that you can create a Mono (version 5.4 and higher) project, and use our assembly from there and run it on Pi Zero. We haven't done a lot of testing on this, but it should theoretically work, and we would be very interested on knowing if that worked for you, so let us know if you give that a try.

alvipeo, 2020-03-03:
Thanks for the info! Should I be able to use `System.Device.Gpio` too?

If that works, it'll be great! I'll give it a try.

joperezr, 2020-03-03:
> Thanks for the info! Should I be able to use System.Device.Gpio too?

Yes 😄 sorry that's what I meant when I said "our library targets .NET Standard 2.0" - by our library I really meant System.Device.Gpio.

alvipeo, 2020-03-04:
This is really great to hear! One thing though - I've never used Mono. Can you please point me out to where to start?

Thank you.

frogcrush, 2020-03-04:
Mono is the open-source .NET Core-(mostly)compatible that came way before .NET Core did. Typically, if you compile for .NET Framework you can run it on .NET Core. [This page](https://docs.microsoft.com/en-us/dotnet/core/about#comparison-with-mono) has some more information.

joperezr, 2020-03-04:
@frogcrush beat me to it, but basically what you want is to create a console app, and set it's targetFramework to `net461`. After that, you should be able to call `dotnet publish -r linux-x64` on it, and it should generate a publish folder with the right dlls and exes in order to make it work on mono linux environment. After installing mono on your linux machine, simply run `mono yourApp.exe` from there, and that should do it.

alvipeo, 2020-03-05:
@joperezr I'm a bit confused. First, how does `net461` relate to .Net Core 3.1? Are you saying this will work?

Secondly, my "linux machine" is Raspberry PI W Zero which has very limited resources (that's why it's extremely cheap and this makes it extremely useful in IoT scenarios!).

joperezr, 2020-03-05:
> @joperezr I'm a bit confused. First, how does net461 relate to .Net Core 3.1? Are you saying this will work?

As @frogcrush pointed out, Mono came before .NET Core. During that time, there were a lot of people that had .NET Framework apps and wanted to be able to run that on linux and OSX. Then came Mono, which the idea behind it was that you could just take your .NET Framework apps and install the mono runtime on your linux machine, and then be able to run them by just saying `mono yourApp.exe`. Latest versions of Mono support .NET Standard 2.0 which is also supported by .NET Framework 4.6.1. One last thing here is that Mono is really just a runtime, not technically a framework, which is why you can't really create a project that targets mono. All this said, it means that you can create a .NET Framework app for 4.6.1, and then use the mono runtime to be able to run that on Linux. Raspberry Pi Zero can't really run .NET Core, but it is able to run the mono runtime, which means that you can use that to run .NET apps on Raspberry Pi Zero. I have tried this in the past by creating Hello world apps and run them in the Pi Zero, but haven't really tried using System.Device.Gpio there which in theory should just work, and I don't have a Pi Zero with me to try so if you could try validating this that would be great. I have already ordered a Pi Zero, so if you don't get a chance of trying I can also give it a try.

alvipeo, 2020-03-05:
got it, thanks! I'll give it a try. Not this week though, too busy.

Ellerbach, 2020-03-14:
@alvipeo, you may want to check this: https://github.com/dotnet/runtime/pull/657 and this: https://github.com/dotnet/runtime/issues/7764
Some work is done to support armv6 for .NET Core. In the mean time, the solution @joperezr gave you will most likely work depending on what you're doing.

Ellerbach, 2020-08-21:
@alvipeo , you will find an example of using the System.Device.Gpio with mono: https://github.com/Ellerbach/Nabaztag.Net
For the bindings, I've been using for this example (see the Nabaztag.Server), I just extracted the code and make it compatible with .NET Standard 2.0. In general, it's about adjusting functions which are using buffers/Span. Nothing really complicated in general.
All up, it's working. So as a summary:
- Create a .NET Framework 4.6.1+ project
- Add the nugget of System.Device.Gpio (it's .NET Standard 2.0 so will be compatible)
- Extract the source code of the bindings you are interested in, adjust the code to be compatible with .NET Standard 2.0 in a specific .NET Standard 2.0 library
- Add from your main project the bindings
- When building the solution from Visual Studio, deploy the full build directory, mono should have all what is necessary to run the project
Et voilà :-) 
Please note that the performance are clearly not as good as for .NET Core but it's working.

krwq, 2020-11-24:
I think we should add some instructions in the repo for Raspberry Pi Zero with mono as part of this issue but other than that we can't really do anything more here. This repo doesn't have control over supported architectures, please upvote the issues in dotnet/runtime to get it more traction.

Caffiendish, 2021-12-23:
Can confirm, Pi Zero 2 works with .NET, I'm using the System.Device.GPIO library, as a .NET 5.0 project IIRC.
The GPIO library took some finagling though, I think I ended up using a nightly for the Pi Zero 2.

alvipeo, 2021-12-23:
I just got my Pi Zero 2 and I will try to make something with .NET 6.0 this weekend.. maybe a bit later ;) But it's very interesting to test it.

Ellerbach, 2021-12-24:
> Can confirm, Pi Zero 2 works with .NET, I'm using the System.Device.GPIO library, as a .NET 5.0 project IIRC. The GPIO library took some finagling though, I think I ended up using a nightly for the Pi Zero 2.

I can confirm Pi Zero 2 W works like a charm with .NET 5.0/6.0 and our .NET IoT lib :-)

alvipeo, 2021-12-24:
Ha! Merry Christmas then! :)

vincenzolauretta, 2022-02-27:
do you guys have a reliable guide to install .NET 5 or 6 to the Pi Zero 2? is there already something like apt-get install net6 or similar that works?

raffaeler, 2022-02-27:
@vincenzolauretta you may want to take a look at this guide: https://docs.microsoft.com/en-us/dotnet/iot/deployment

In this thread you may also find some comments on the pros/cons of the different deployment methods: https://github.com/dotnet/iot/issues/1783

joperezr, 2022-02-27:
@vincenzolauretta installing dotnet globally with a package manager depends on the OS you have installed. Here is a link with the [general instructions](https://docs.microsoft.com/en-us/dotnet/core/install/linux?WT.mc_id=dotnet-35129-website) on how to install with package manager, and assuming you installed Raspbian on the Pi, you can use the [Debian instructions](https://docs.microsoft.com/en-us/dotnet/core/install/linux-debian) to install it via apt-get

krwq, 2023-01-19:
[Triage] Closing this given the support for the old ARM is outside of dotnet/iot control.

closed 2023-01-19 by krwq