dotnet/iot #1010: iot/src/devices/Dhtxx/ Issues

djaus2, 2020-03-14:
**Describe the bug**

DHTxx running in 1-wire mode with DHT22 misses some conversions. Also runs on Raspian on the RPi but blocks with no response to Dht22 instantiation on IoT-Core.

**Steps to reproduce**
I've create a sampler app on [GitHu djaus2/DNETCoreGPIO](https://github.com/djaus2/DNETCoreGPIO). 
```csharp
See code in repository as above.
```

**Expected behavior**

Expect to see temperature and pressure displayed.

**Actual behavior**

On Raspian some conversions report failure. Most succeed.
On the IoT-Core doesn't get past Dht22 instantiation.

**Versions used**
I am using the Nuget Packagaes
```csharp
  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.1</TargetFramework>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="Iot.Device.Bindings" Version="1.1.0-prerelease.20153.1" />
    <PackageReference Include="System.Device.Gpio" Version="1.1.0-prerelease.20153.1" />
  </ItemGroup>
```csharp

Add following information:

- `dotnet --info` on the machine being used to build
[192.168.0.11]: PS C:\share\DNETCoreGPIO> dotnet --info
.NET Core SDK (reflecting any global.json):
 Version:   3.1.101
 Commit:    b377529961

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.17763
 OS Platform: Windows
 RID:         win10-arm
 Base Path:   c:\Quickstarts\dotnet\sdk\3.1.101\

Host (useful for support):
  Version: 3.1.1
  Commit:  a1388f194c

.NET Core SDKs installed:
  3.1.101 [c:\Quickstarts\dotnet\sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.1.1 [c:\Quickstarts\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.1 [c:\Quickstarts\dotnet\shared\Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
- Version of `System.Device.Gpio` package .. As above
- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`) .. As above

djaus2, 2020-03-14:
Note that the GPIO functionality in the Sampler works on IOT-Core.

djaus2, 2020-03-15:
Further:
- I tried 1k and 10k ohm pullups with no impact upon the missed conversions.
- Making the time between conversions longer also had no impact on that. Nb The Dht22 specs say min 2 secs between.
- I ran the DHT22 on an Arduino Uno with no conversion problems. That code does check for nans.

djaus2, 2020-03-15:
Looking at the sample UWP code at for example:
[hackster.io](https://www.hackster.io/porrey/go-native-c-with-the-dht22-a8e8eb) it is stated that IoT-Core can't change the pin mode fast enough and therefore requires 2 pins, one in, out and a MSFET to control the in out on 1-wire from/to the sensor.  

So am I correct in concluding that the DHT22 driver here as in the sample won't work on Win10 IoT-Core?

joperezr, 2020-03-16:
> So am I correct in concluding that the DHT22 driver here as in the sample won't work on Win10 IoT-Core?

This I'm not sure is true or not. @Ellerbach have you ever tried running the sensor on IoT-Core or just Raspbian? 

> On Raspian some conversions report failure. Most succeed.

@djaus2 I'm not sure what you mean by this failure, it would be great if you could provide more details like exception message or error call stack so that we could investigate more. Also, there was a recent fix on this sensor merged late-last week https://github.com/dotnet/iot/pull/1008, so you might want to give that a try and see if you still see the errors on Raspbian. The last comment I added on that PR will show you the instructions on how to get a nightly build of the package that already has this fix in.

Ellerbach, 2020-03-17:
@joperezr 
I never tried on Windows IoT Core. I will try as soon as I can. I'm always using Raspbian for this. My guess is that on the Windows platform, we will have hard time to manage the OneWire protocole. And yes, a very old hack (from the .NET Microframework time) was to use 2 pins in this case exactly as described by @djaus2. So in the case, the OneWire will not be supported properly on Windows platform, we should then either use the hack either add in the doc that the platform is not supported for OneWire all up.

djaus2, 2020-03-17:
@joperezr @Ellerbach 
Raspian IS OK. I just trap the failed conversions.  About 1 in 10, sometimes more frequent.
On IoT-Core it does not get past the instantiation of the class.  Just hangs there.
I will try an update on the package.

Yeah might be worth documenting that 1-wire isn't supported on IOT-Core

Ellerbach, 2020-03-18:
So I ran tests and seems like as expected, the OneWire is not working properly on Windows IoT Core. Switching state is not fast enough.
And yes, the trick would be to use 2 pins. This trick was already apply in an older version on Windows 10 IoT Core as well: https://github.com/ms-iot/samples/tree/develop/GpioOneWire.
I would prefer to mark that OneWire protocol is not working properly on Windows 10 IoT Core rather than having 2 wires just for Windows 10 IoT Core. The protocol would not be called anymore OneWire but TwoWires :-)

joperezr, 2020-03-18:
@Ellerbach would either you or @djaus2 be able to add few lines on the README of DHT to say that OneWire doesn't work well in Windows, and perhaps some instructions on how to set it up with Two Wires? Perhaps  after doing that we can then close this issue as documentation for now and ping the WINRT team about this to see if they can do anything to improve this scenario in Windows.

djaus2, 2020-03-18:
Ok I will do it seeing I started this.

David Jones MVP

pgrawehr, 2020-03-18:
The performance issues could be related to #1004. A short investigation showed that switching pin levels is just very slow on Windows for some reason.

djaus2, 2020-03-24:
Conclusion is that **1-wire will not work with Win 10 IoT-Core** as the OS cannot switch the pin fast enough between output and input (or viz). There are some fixes which use two GPIO pins. One for input and one for output., but that then would not be 1-wire. 
Nb: The 2-wire "fix"  is discussed here: [GPIO OneWire DHT11/22 reader](https://github.com/Microsoft/Windows-iotcore-samples/tree/develop/Samples/GpioOneWire) and here: [hackster.io](https://www.hackster.io/porrey/go-native-c-with-the-dht22-a8e8eb)

closed 2020-03-24 by djaus2