dotnet/iot #1018: Support for MIMXRT1060-EVK

ca-t, 2020-03-20:
Can I build applications for [MIMXRT1060-EVK](https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/mimxrt1060-evk-i-mx-rt1060-evaluation-kit:MIMXRT1060-EVK) ?

Or any plans to add it?

[image]

joperezr, 2020-03-23:
mm good question, we actually don't have one to try, but do you have any additional info on what processor does your PCB have? Or more specifically, have you tried running .NET Core on it? The answer in terms of System.Device.GPIO library, is that as long as your PCB can run a framework that supports .NET Standard (either .NET Core, or Mono mainly) then you will be able to use our library on it.

ca-t, 2020-03-23:
Processor is [MIMXRT1062DVL6A](https://www.nxp.com/part/MIMXRT1062DVL6A#/)
Here is [Datasheet Link](https://www.nxp.com/docs/en/nxp/data-sheets/IMXRT1060CEC.pdf)

[image]

I want to run .NET Core but I've not figured out all details yet. I guess I need to build own image for this board and flash it. And then I can deploy dlls.

joperezr, 2020-03-23:
I see, for now what you could try is to just run a hello world app in there. To do that, build your app in Windows (or whatever your dev box) by doing:

```
dotnet new console -n SampleHelloWorldApp
cd SampleHelloWorldApp
dotnet publish -r linux-arm
```

That will generate a publish folder that has all the files you need to copy over to your PCB in order to run the app. I'm also not sure if Arm M7 is 64-bit, if so, then change your publish command to instead do `dotnet publish -r linux-arm64`

After copying everything to your PCB, you should be able to run them by

```
cd <whereverYouCopiedThePublishFolderContents>
chmod +x SampleHelloWorldApp
./SampleHelloWorldApp
```

Let us know if by doing that, you see the Hello World app running succesfully.