dotnet/iot #1020: Waveshare Ssd1351 issues

techniq, 2020-03-23:
**Describe the bug**

I am using a [Wareshare RGB OLED](https://www.amazon.com/1-5inch-RGB-OLED-Module-Communicating/dp/B07DB5YFGW/ref=sr_1_1?keywords=ssd1351&qid=1584921798&sr=8-1) with the Ssd1351 driver chip instead of an [Adafruit board](https://www.adafruit.com/product/1431) (it is about 1/2 the price) but I'm having trouble getting it to work with the dotnet/iot [driver](https://github.com/dotnet/iot/tree/master/src/devices/Ssd1351).

I've been able to successfully run Waveshare's demo code (in Python) so I've confirmed it's wiring is correct (I updated the Python demo to use the same RST / DC / CS pins as the dotnet sample).  I've also attempted to match the Python initialization commands with what is being done in `dotnet/iot` with a few tweaks which are about 98% identically, but I still haven't had any success yet.

I was hoping someone (original driver writer @Frankenslag possibly) might be able to spot a difference between the boards (see manual) or what is being done in the Python demo code vs. `dotnet/iot`.  I can post the changes I've tried if that would help (timings in `ResetDisplayAsync`, commands used in `InitDisplayForAdafruit`)

- [Waveshare User Manual](https://www.waveshare.com/w/upload/5/5b/1.5inch_RGB_OLED_Module_User_Manual_EN.pdf)
- [Wiki with Demo Code](https://www.waveshare.com/wiki/1.5inch_RGB_OLED_Module) (Python or C)

**Steps to reproduce**

Attempt to run the [Ssd1351 Sample](https://github.com/dotnet/iot/tree/master/src/devices/Ssd1351/samples) using a [Waveshare RGB OLED](https://www.amazon.com/1-5inch-RGB-OLED-Module-Communicating/dp/B07DB5YFGW/ref=sr_1_1?keywords=ssd1351&qid=1584921798&sr=8-1&th=1) board (or possibly another non-Adafruit board) such as searching [Amazon](https://www.amazon.com/s?k=ssd1351) for Ssd1351.

**Expected behavior**

Sample images are displayed

**Actual behavior**

No output on display (but works with Python demo)

**Versions used**

Add following information:

- Version of `System.Device.Gpio` package - `1.0.0` and `master`
- Version of `Iot.Device.Bindings` package - `1.0.0` and `master`

Frankenslag, 2020-03-23:
@techniq 
Comparing the code then the Ssd1351 driver uses Spi Mode3 and looking at the timing diagrams this would appear correct `CPOL 1, CPHA 1. Polarity is idled high and data is sampled on rising edge of the clock signal.`

The sample for the Waveshare uses Mode0 and I am not sure why. Is there an inverter on the board ?

Anyway what happens if you modify the sample to use Mode0 ?

techniq, 2020-03-23:
@Frankenslag That was the trick.  Thanks!

Changed:
```c#
using (SpiDevice displaySPI = SpiDevice.Create(new SpiConnectionSettings(0, 0) { Mode = SpiMode.Mode3, DataBitLength = 8, ClockFrequency = 12_000_000 /* 12MHz */ }))
```

To:
```c#
using (SpiDevice displaySPI = SpiDevice.Create(new SpiConnectionSettings(0, 0) { Mode = SpiMode.Mode0, DataBitLength = 8, ClockFrequency = 12_000_000 /* 12MHz */ }))
```

I still need to back out some of my other changes (I was waiting `500ms` in `ResetDisplayAsync()` instead of just `20ms` to match their example, as well as some tweaks to the order of commands in `InitDisplayForAdafruit`).  I'm still getting some intermittent issues but re-running the sample usually fixes it.

This is my first SPI device so I'm still learning :).  Thanks again.

closed 2020-03-23 by techniq

techniq, 2020-03-23:
@Frankenslag Btw, I'm not sure about the inverter, but here is a picture of the back if that helps :)

[image]

There is a resistor to to change from 4-wire to 3-wire SPI.