dotnet/iot #1025: Improve the coding guidelines document

andrasfuchs, 2020-03-25:
The documentation about the coding guidelines are very minimal, and the link to the corefx guidelines is broken. 

See it here:
https://github.com/dotnet/iot/blob/master/Documentation/Coding-guidelines.md

I'm trying to contribute by adding a driver for a new device (#1024), and I spent the last 2 hours fixing formatting problems because of the fairly strict rules. 
After every new commit I had to wait for the PR's compilation to see the next 60 or so build problems (because I have different rules at the moment locally).

Could you publish your coding guideline ruleset so that I can verify my code within Visual Studio before checking it in?

krwq, 2020-03-25:
cc: @pgrawehr

@andrasfuchs if you build using build.cmd/build.sh it should happen automatically.
Ruleset can be found here: https://github.com/dotnet/iot/blob/master/eng/StyleCop.Analyzers.ruleset

pgrawehr, 2020-03-25:
Yes, you should see the rules being automatically applied when building using the build script or in Visual Studio. I immediatelly get red underlines everywhere where I break a rule, often even with a suggestion on how to fix it. Which tools are you using to write code and build?

andrasfuchs, 2020-03-26:
I haven't used the build script yet.

I have Visual Studio 2019 v16.4.2. I think I got it working, here is what I did:

1, Added the StyleCop.Analyzers NuGet package to my project
2, Put [the above mentioned StyleCop ruleset](https://github.com/dotnet/iot/blob/master/eng/StyleCop.Analyzers.ruleset) into my projects root folder
3, Added the following line to my .csproj file's PropertyGroup section:
`<CodeAnalysisRuleSet>StyleCop.Analyzers.ruleset</CodeAnalysisRuleSet>`
4, Got [the recommended .editorconfig](https://github.com/dotnet/runtime/blob/master/.editorconfig) from dotnet/runtime repo, and put into the root folder of my solution

Now all the formatting warnings are actual errors, and the rules look very similar to the ones I got from GitHub's compiler with my PR.

Is this the right way to do this?

pgrawehr, 2020-03-26:
This works, but normally, this should not be necessary, since the StyleCop configuration is automatically loaeded from the `Directory.Build.props` file in the project root directory (and the src/devices/ dir). Do you have any .props or .targets files in your binding directory (src/devices/AD7193/)? Remove them. Also, you haven't commited the AD7193.csproj file, if I saw correctly - maybe there's something else wrong with that.

andrasfuchs, 2020-03-26:
I removed the .props and .targets from my obj directory.

Alright, I'm totally confused at the moment. It looks like the .editorconfig caues more trouble, because it has contradictory rules with the StyleCop .ruleset. If I put the usings after the declaration of the namespace for example, one of them generates and error, if I put it outside of that, the other generates an error.

I created the Ad7193.csproj file too, but either I get an error that the Device namespace is not in the System namespace (missing project reference maybe?), or if I add the System.Device.Gpio as a project reference, then I get duplicate errors for SpiMode existing in both libraries, while the two libraries listed look the same.

> CS0433	The type 'SpiMode' exists in both 'System.Device.Gpio, Version=42.42.42.42, Culture=neutral, PublicKeyToken=31bf3856ad364e35' and 'System.Device.Gpio, Version=42.42.42.42, Culture=neutral, PublicKeyToken=31bf3856ad364e35'	Ad7193	C:\Work\dotnet\iot\src\devices\Ad7193\Ad7193.cs

Do you have a written documentation that I can follow step by step as a first time contributor? I'm ok if I have to recreate my project and/or references, I just want to get my PR ready as quickly as possible, and go on with my project.

pgrawehr, 2020-03-26:
There are things a bit tricky here, due to the build setup for different platforms. It is planned to simplify that in the future, but for now it is not really easy (I'm constantly fighting with problems like the ones you describe as well). 
Give me a few minutes to check what's wrong on your branch.

pgrawehr, 2020-03-26:
Ok, I've got a patch for you. Just apply it and use the new solution file that's included. When building in VS, make sure that you choose either "Windows-Debug" or "Linux-Debug" (or "...-Release") as the Solution configuration. This is a workaround to reference the correct version of System.Device.Gpio for the respective platform. 

The build passes for me, both from VS and from the command line. The main thing you still need to do is remove all the `#pragma warning disable CS1591` lines I've added and provide meaningful documentation.

@joperezr Have a look at the project files, please. I'm just testing to change all project files with such a conditional include, because now some of the samples don't run under windows, even if they should. Especially many unit tests are not executed on Windows, just because RuntimeIdentifier is always set to Linux for the System.Device.Gpio reference.

andrasfuchs, 2020-03-26:
Thank you @pgrawehr , I really appreciate it. I'll try it first thing tomorrow.

andrasfuchs, 2020-03-27:
I applied the patch and it worked fine, thank you! I also removed all `#pragma warning disable CS1591` lines and added the XML documentation as you recommended, and checked in the changes.

It finally build on Windows and MacOS, which is great news, but I'm not sure if it's normal that the Linux build is still in progress after 46 minutes.

[image]

pgrawehr, 2020-03-27:
This happens sometimes, it is not your fault. It will eventually fail after an hour. Do a dummy commit (or force push) to start the build again.

joperezr, 2020-03-27:
@andrasfuchs first of all thanks for your contribution. We actually had network infrastructure issues in our build lap which has all the hardware that runs these tests which is causing your build to hang. That is resolved now, so if your job didn't go and finish let me know so that I can trigger a re-run.

joperezr, 2020-03-27:
>I haven't used the build script yet.

@andrasfuchs because of the way that our repository builds (and as @pgrawehr explained above) you do have to run the scripts first before opening any project in VS. I bet that most of the issues you are currently hitting are due to not running the build scripts first, but once you do you should get the stylecop warnings in VS as well on your new projects without the need of adding new package dependencies or stylecop rules yourself. If you hit any issues while doing that, please do let me know as we definitely want to fix that.

andrasfuchs, 2020-03-28:
@joperezr Thank you, in this case, could you trigger the re-run please?

I run build.cmd now, but I can't confirm that it would have helped in my case, because I already applied the above patch, and it worked. 

I just re-read the contributing.md you have in this repo, and it says it clearly to run it first, so sorry, it was my fault that I didn't follow the instructions properly.

closed 2020-03-31 by andrasfuchs