dotnet/iot #1027: Bmp280 Sensor gives wrong readings

pgrawehr, 2020-03-26:
While testing the Bmp280 sensor, I found that mine was giving very wrong readings (~940hPa when it should be in the order of ~1010hPa). There's definitelly no storm outside now, and it's cold but sunny. 

Investigating further, I found a small bug in the calibration calculation for temperatures: 

Line 181 of Bmxx80Base.cs should be changed to: 
```
double var1 = ((((double)adcTemperature / 16384.0) - ((double)_calibrationData.DigT1) / 1024.0)) * (double)_calibrationData.DigT2;
```            
There are missing parentheses in that formula. 

I'll come up with a patch for this (including an unit test), however, since my sensor still gives the wrong values (also with the python reference implementation), it's hard for me to know that's really all. 

See also #1022

CC @djaus2

djaus2, 2020-03-26:
Here are my comparisons, without making the changes as above:
DMP180 on RPI-Raspian: 1014
DMP180 on RPI-IoT Core: 1014
DMP180  on Arduino Uno: 1022  (same sensor)
Local Weather Station (2km away approx same elevation): 1026

So for me the .Net Core version is only a little off.

joperezr, 2020-03-31:
That is a great find @pgrawehr, please feel free to submit a PR to fix this when you can 😄

pgrawehr, 2020-04-01:
Since my sensor is apparently broken (or is incorrectly calibrated) I would be glad if @djaus2 could verify the fix. I have ordered a new one, but I'm not sure when this arives.

djaus2, 2020-04-01:
Ok will do in next 24hrs

David Jones MVP

djaus2, 2020-04-02:
Haven’t got to this yet…coming.

pgrawehr, 2020-04-08:
I found the culpit (after the second, brand new sensor gave the exact same readings...). The sensor delivers absolute values at the place of measurement, but to get the equivalent barometer reading, one needs to correct this for the altitude at the point of measurement. Since I'm at 650m over sea, the difference is significant already. Applying a correction formula helped. So part of the problem is that #783 still is not merged.

djaus2, 2020-04-09:
Looking at this now. I now have a BME280

djaus2, 2020-04-10:
Got my BME280 going. On Arduino, the results were unreliable.
Using repository here unchanged (on IoT-Core) I got:

Temperature:   20.22040450561917 °C | Pressure: 1008.5941796875001 hPa | Altitude: 38.83519682937275 meters | Humidity: 52.66412078332605 %
-- | -- | -- | --
Temperature:   20.217536038577805 °C | Pressure: 1008.5810156250001 hPa | Altitude: 38.945206702271896 meters | Humidity: 52.472211632850716 %
Temperature: 20.21785475713341   °C | Pressure: 1008.5882421875 hPa | Altitude: 38.88481538182217 meters | Humidity: 52.45529727652814 %
  |   |   | ^ Inside
Local: 18.4 | Local: 1016 | GPS says: 33m | Local: 71
Themometer:  19.9 | Sea Level about 15km away: 1015 |   |

djaus2, 2020-04-10:
I made the brackets change Line 181 of Bmxx80Base.cs
Reults now:

Temperature:   20.630277520627715 °C | Pressure: 1008.0932421875 hPa | Altitude: 43.022267781604704 meters | Humidity: 52.97567954946546 %
-- | -- | -- | --
Temperature: 20.650994324802582 °C | Pressure: 1008.05703125 hPa | Altitude: 43.32500109430068 meters | Humidity: 52.96501889587134 %

djaus2, 2020-04-10:
With the Line 181 of Bmxx80Base.c:

Temperature:   17.954344242107254 °C | Pressure: 1004.8240625 hPa | Altitude: 70.5845337247005 meters | Humidity: 40.167819101323936 %
-- | -- | -- | --
Temperature: 17.939683561131915 °C | Pressure: 1004.804375 hPa | Altitude: 70.55408235285238 meters | Humidity: 40.15090910070303 %
Thermometer 16 |   |   |  
  | Nb: App gives Sealevel Pressure as : 1013.25 |   |

pgrawehr, 2020-04-12:
So the effect of the fix is marginal, but I'll later push it anyway (together with an unit test for the Bmp280 calibration formula). 
My main problem was actually the missing height reduction formula. I've just pushed an updated WeatherHelper PR that gives values more in the range of what I expect.

djaus2, 2020-04-12:
NB: I'm currently doing a port of DME280 to Azure Sphere based upon Seeed's version.

djaus2, 2020-04-14:
Got the DME280 working OK on the Az Sphere. Results are consistent with what I got as aboe.

joperezr, 2020-05-07:
Nice! I suppose you mean that is with the line that @pgrawehr suggested above. If that's the case can either you or @pgrawehr put up a PR to fix this issue on the binding?

pgrawehr, 2020-05-07:
Actually, I'm a bit confused, because I think I didn't really change anything (the parentheses on that line are extra-confusing). But I've done it together with an unit test and an improved calcuation formula (using doubles instead of fixed-point math). I can finish that PR pretty quickly, but I don't want to have to many PRs open at a time.

PR #1073 (2020-07-02): Bmp280 test
Fixes #1027

Reading is apparently correct now. Added self test for math formula and extended example a bit to show some weatherhelper usage.
closed 2020-07-02 by joperezr