dotnet/iot #1031: Add support for alarms on the DS3231 RTC

henryshunt, 2020-03-30:
The DS3231 RTC implementation (in devices/Rtc) at present only includes functionality to get the time, set the time and get the temperature. The device itself also has the ability to set two separate alarms that will fire when the specified time is matched (according to various matching modes), which I am in need of for a project I am working on.

[This](https://github.com/Makuna/Rtc/) repo, which I have used in other projects, provides a C implementation of the DS3231 (including alarms). The relevant parts for alarms in that repo (see src/RtcDS3231.h) are anything to do with alarm one/two, square wave, and latch. Note: I'm not sure how the license of that repo affects the ability to utilise it in whatever way here, but I thought I'd reference it just to show what needs implementing.

Personally I only need alarm one working with the once-per-second mode.

joperezr, 2020-03-31:
Cc: @ZhangGaoxing who I believe originally added the binding.  The repo you provided seems to be under GPL license which would mean we wouldn’t be able to look at that source. We could probably find other repos with a more permissive license that we could use. @henryshunt would you be interested in adding this support for this binding?

henryshunt, 2020-04-01:
I've never done anything like this before, but I've had a look at the datasheet, the current code, and attempted to write some of the required code and I think I should be able to do it. So yes, I'll work on adding the support for alarms.

joperezr, 2020-04-02:
Awesome, there is no rush and no pressure, and I believe that this looks to be a good first issue to start with if you are up to it. I'm happy to help you through out the way, feel free to reach out if you hit any issues. Do remember that once you fork the code, first thing you need to do is to run build.cmd/build.sh script from the root of the repo, after that you should be able to open the device's project in VS (or whatever editor you prefer) and start your dev work 😄

henryshunt, 2020-04-03:
I actually got alarm 1 of 2 working just yesterday so I shouldn't have any problems getting this done.

But I'm unable to open the Rtc project in VS. It comes up with this error:
The project file cannot be opened by the project system, because it is missing some critical imports or the referenced SDK cannot be found.
Detailed Information: Unable to locate the .NET Core SDK. Check that it is installed and that the version specified in global.json (if any) matches the installed version.

It seems the problem is that the standard VS .NET Core workload installs 3.1.101 but the repo requires 3.1.201. It is okay to just temporarily change the SDK version in global.json to 201 (everything works fine then) or should I install 101 manually?

Also, when I close the project it wants to save a solution (which doesn't exist). I assume I should not commit that file?

joperezr, 2020-04-09:
I believe you are right and the issue you are getting is because you don't have the SDK we require installed. 3.1.101 is a bit outdated (about 3 months old now) so I would recommend you installing the latest on your machine independent of the requirement from our repo. Assuming you have windows X64 machine, here is the link to install that: https://dotnet.microsoft.com/download/dotnet-core/thank-you/sdk-3.1.201-windows-x64-installer or if you have a different platform, you can find all .NET Core downloads at http://get.dot.net/

joperezr, 2020-04-09:
As for the solution, yes it is fine to either not save the solution when you close, or to save it but not commit it. It is actually very likely that we will end up adding solutions for every device anyways soon.

PR #1150 (2021-01-21): Update DS3231 binding to add support for the alarms
Fixes #1031 

Adds support for both of the built-in alarms on the DS3231 RTC chip.
closed 2021-01-21 by Ellerbach