dotnet/iot #1036: Many Unit tests don't execute on Windows

pgrawehr, 2020-04-04:
**Describe the bug**

If I execute
```
C:\projects\iot4>build -test
```
on Windows, the last few lines of the output read

```
  Mcp23xxx.Tests -> C:\projects\iot4\src\devices\Mcp23xxx\tests\bin\Debug\netcoreapp2.1\Mcp23xxx.Tests.dll
  Mcp25xxx.Tests -> C:\projects\iot4\src\devices\Mcp25xxx\tests\bin\Debug\netcoreapp2.1\Mcp25xxx.Tests.dll
  Pcx857x.Tests -> C:\projects\iot4\src\devices\Pcx857x\tests\bin\Debug\netcoreapp2.1\Pcx857x.Tests.dll
Testlauf für "C:\projects\iot4\src\devices\Pcx857x\tests\bin\Debug\netcoreapp2.1\Pcx857x.Tests.dll" (.NETCoreApp,Version=v2.1)
Microsoft (R) Testausführungs-Befehlszeilentool Version 16.3.0
Copyright (c) Microsoft Corporation. Alle Rechte vorbehalten.

Die Testausführung wird gestartet, bitte warten...

Insgesamt 1 Testdateien stimmten mit dem angegebenen Muster überein.

Der Testlauf war erfolgreich.
Gesamtzahl Tests: 20
     Bestanden: 20
 Gesamtzeit: 1.1058 Sekunden
  ServoMotor.Tests -> C:\projects\iot4\src\devices\ServoMotor\tests\bin\Debug\netcoreapp2.1\ServoMotor.Tests.dll
  Ssd13xx.Tests -> C:\projects\iot4\src\devices\Ssd13xx\tests\bin\Debug\netcoreapp2.1\Ssd13xx.Tests.dll
  Units.Tests -> C:\projects\iot4\src\devices\Units\tests\bin\Debug\netcoreapp2.1\Units.Tests.dll

Der Buildvorgang wurde erfolgreich ausgeführt.
    0 Warnung(en)
    0 Fehler

Verstrichene Zeit 00:00:42.88
```
So apparently only the unit tests for Pcx857x were executed, but not the others. I could not see any relevant difference ie. between Pcx857x.Tests.csproj and ServoMotor.Tests.csproj. It also doesn't seem to be related to some wrong naming, as explicitly specifying the ServoMotor.Tests.csproj in `build.proj`, line 24, doesn't help (but explicitly specifying an inexistent file does throw an error). Also, the ServoMotor tests execute successfully from within VS. 

**Steps to reproduce**

```
run "build -test" on the command line on Windows
```

**Expected behavior**

All unit tests execute.

**Actual behavior**

Only the unit tests for Pcx857x are executed

**Versions used**

Latest master branch.

pgrawehr, 2020-04-06:
Not 100% sure, but looks like it doesn't work on linux as well.

joperezr, 2020-04-06:
This is happening because of issue https://github.com/dotnet/iot/issues/798 which is actually a bug with VSTest: https://github.com/microsoft/vstest/issues/2224

Essentially what happens here is that VSTest is incorrectly using Console.WriteLine instead of using the MSBuildLogger, and what this does is that if the build is multi-threaded (which ours is) only the thread that has grabbed the console will print out the results, but the rest of the threads would just be running Console.WriteLine to output wihtout actually printing anything.

joperezr, 2020-04-06:
Closing as we already have an issue in this repo tracking this external fix.

closed 2020-04-06 by joperezr