dotnet/iot #1040: System.PlatformNotSupportedException: Operation is not supported on this platform.

gtrztu, 2020-04-10:
Hello, I am getting into a real issue when I am trying to run my .dll console app with mono in Raspbian 10 on my Raspberry Pi 4B. The error that occurs is when the constructor for the GpioController is reached in the code I think, because I have got string output before that and it's ok. The package System.Device.Gpio was successfully implemented to the project before the build... so... I am begging for help!!

Unhandled Exception:
System.PlatformNotSupportedException: Operation is not supported on this platform.
  at ca2.Program.Main (System.String[] args) [0x0000c] in <24b92db2ff6f47f0918e4ec97b04a401>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.PlatformNotSupportedException: Operation is not supported on this platform.
  at ca2.Program.Main (System.String[] args) [0x0000c] in <24b92db2ff6f47f0918e4ec97b04a401>:0

pgrawehr, 2020-04-10:
This information is quite vague, because the callstack you show does not show any function of the library, but just your main routine. Can you show the code you use to construct the GpioController?

gtrztu, 2020-04-10:
Absolutely simple code...

`Console.WriteLine("running...\n");
            GpioController controller = new GpioController(PinNumberingScheme.Board);
            controller.OpenPin(6, PinMode.Output);
            while (true)
            {
                controller.Write(6, PinValue.High);
                Thread.Sleep(100);
                controller.Write(6, PinValue.Low);
                Thread.Sleep(100);
            }`

pgrawehr, 2020-04-11:
Looks a bit like you got the library for the wrong platform. You need to make sure you build for linux to run on the Raspberry Pi. It's easiest to use the "Publish..." feature in Visual Studio (on your application's project) to bundle the correct libraries. Make sure the target platform is set as linux-arm. 

You can try to use `GpioController controller = new GpioController(PinNumberScheme.Board, new RaspberryPi3Driver())` to make sure the correct library is included.

joperezr, 2020-04-11:
As @pgrawehr suggests here, the problem is the way you are publishing your application since you are getting the platform independent implementation which just throws platform not supported. In order to publish it correctly for mono on raspbian, try running `dotnet publish yourProject.csproj -f net461 -r linux-x64` that should put the right implementation on your output folder and get past the error you are hitting.

pgrawehr, 2020-04-11:
@joperezr I believe this should be `dotnet publish yourProject.csproj -f net461 -r linux-arm`. The raspberry has no x64 CPU 😄

joperezr, 2020-04-11:
You are correct 😀

gtrztu, 2020-04-11:
Thank you so much, but I followed the suggestions and now I get the error after the PowerShell line `dotnet publish ca4.csproj -f netcore3.1 -r linux-arm`
error:
`C:\Program Files\dotnet\sdk\3.1.201\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.TargetFrameworkInference.targets(93,5): error NETSDK1013: The TargetFramework value 'netcore3.1' was not recognized. It may be misspelled. If not, then the TargetFrameworkIdentifier and/or TargetFrameworkVersion properties must be specified explicitly. [C:\Users\Galin\Desktop\IoTshower\ca4\ca4\ca4.csproj]`

pgrawehr, 2020-04-12:
My bad. It should be "netcoreapp3.1".

joperezr, 2020-04-12:
If you want to run on Mono, then the framework of your project should be a .NetFramework one which is why in my instructions I said to use is net461. The only thing wrong with my instructions was that for the -r argument I said to use linux-x64 but it should be linux-arm instead. If you build against netcoreapp you wont be able to run on Mono

gtrztu, 2020-04-12:
Ohh… I'm sorry for the dummy problems but I am getting more and more errors... Now I made .NETFramework4.7.2 project with the GPIO package and in VS everything is ok...no errors. But in PowerShell after the fixed line `dotnet publish final2.csproj -f net472 -r linux-arm` I am getting new type error... It is:

`C:\Program Files\dotnet\sdk\3.1.201\Microsoft.Common.CurrentVersion.targets(3032,5): error MSB4216: Could not run the "GenerateResource" task because MSBuild could not create or connect to a task host with runtime "CLR4" and architecture "x86".  Please ensure that (1) the requested runtime and/or architecture are available on the machine, and (2) that the required executable "C:\Program Files\dotnet\sdk\3.1.201\MSBuild.exe" exists and can be run. [C:\Users\Galin\Desktop\IoT\final2\final2\final2.csproj]`

And next these lines a thousand lines with errors in white color... I'm feeling so stuck...

gtrztu, 2020-04-12:
The error above was for windows forms app, because this is the thing I am trying exactly to do. Now I am trying with console application with the same PowerShell line and I get the error :

`C:\Program Files\dotnet\sdk\3.1.201\Microsoft.Common.CurrentVersion.targets(3947,5): error MSB4096: The item "C:\Users\Galin\Desktop\IoTshower\ca5\packages\System.Runtime.WindowsRuntime.4.7.0\build\net461\..\..\ref\netstandard2.0\System.Runtime.WindowsRuntime.dll" in item list "ReferencePath" does not define a value for metadata "CopyLocal".  In order to use this metadata, either qualify it by specifying %(ReferencePath.CopyLocal), or ensure that all items in this list define a value for this metadata. [C:\Users\Galin\Desktop\IoTca5\ca5\ca5.csproj]`

I just want a form that runs c# code and the gpio pins on my Rpi 4...

joperezr, 2020-04-12:
Oh, winforms doesn’t work on linux, even when using Mono (AFAIK). You will only be able to do it for a console app. Also, why do you want to use Mono? .NetCore runs on RPi4 and our library has been tested against that runtime, so I would recommend writing a .net core console app to control gpio pins

gtrztu, 2020-04-12:
Thank you so much. Is it possible to program a GUI that works on Linux and uses c# to control the GPIO on Rpi? What is your recommendation?

joperezr, 2020-04-12:
There are many open source .Net projects that provide cross platform UI, but I haven’t tried many so I wouldn’t be able to recommend one over the other. I’ve only developed a couple of apps for the Pi where I wanted to have some UI, and in those cases I developed an ASP.NET MVC site fore the UI running on the PI which would internally call gpio via System.Device.Gpio

Ellerbach, 2021-07-01:
[Triage] This has been solved now for some time.

closed 2021-07-01 by Ellerbach