dotnet/iot #1042: Ws2812b binding incorrect light color

DanielSSilva, 2020-04-14:
**Describe the bug**
The [Ws28xx sample](https://github.com/dotnet/iot/blob/master/src/devices/Ws28xx/samples/Ws28xx.Sample.cs) is not displaying some colors correctly, namely Red. Changed [this sample](https://github.com/dotnet/iot/blob/master/src/devices/Ws28xx/samples/Ws28xx.Sample.cs#L34-L44) to the following

**Steps to reproduce**

```csharp
img.Clear();
img.SetPixel(0, 0, Color.Red);
neo.Update();
```

but instead of displaying red, it's displaying purple

**Expected behavior**
The Led should show a red light

**Actual behavior**
A purple light is being shown

**Versions used**

Add following information:

- `dotnet --info` on the machine being used to build

```.NET Core SDK (reflecting any global.json):
 Version:   3.1.201
 Commit:    b1768b4ae7

Runtime Environment:
 OS Name:     ubuntu
 OS Version:  18.04
 OS Platform: Linux
 RID:         ubuntu.18.04-x64
 Base Path:   /usr/share/dotnet/sdk/3.1.201/

Host (useful for support):
  Version: 3.1.3
  Commit:  4a9f85e9f8

.NET Core SDKs installed:
  2.1.805 [/usr/share/dotnet/sdk]
  2.2.402 [/usr/share/dotnet/sdk]
  3.1.201 [/usr/share/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.1.17 [/usr/share/dotnet/shared/Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.8 [/usr/share/dotnet/shared/Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.17 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.8 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.3 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.17 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.8 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.3 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
```

- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
- Version of `System.Device.Gpio` package
I've cloned the repo and changed directly, then copied the output to my raspberryPi. PowerShell says:
```powershell
PS /home/pi/testLeds_dotnet> [System.Diagnostics.FileVersionInfo]::GetVersionInfo('./System.Device.Gpio.dll').FileVersion
42.42.42.42424
```

- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)

krwq, 2020-04-16:
@DanielSSilva I've used this library before without any issues. Could you double check your device is definitely WS2812b and that you changed your `/boot/config.txt` to have following:

```
dtparam=spi=on
core_freq=250
core_freq_min=250
```
as per https://github.com/dotnet/iot/tree/master/src/devices/Ws28xx

krwq, 2020-04-21:
@DanielSSilva have you perhaps had a chance to see if the above helped?

DanielSSilva, 2020-04-21:
Sorry @krwq I've forgot to answer.
I can't be 100% sure that those are ws2812b, but I'm 99% sure since I've used different libraries for such model and all worked.
Regarding the configurations, I've set that as the wiki specifies.

On a side note, the other libraries that I've used, I think that they didn't use the SPI, not sure though

DanielSSilva, 2020-04-21:
Also, I can't remember from the top of my head, but there are other colors that are being wrongly represented.
Funny enough, dark red shows as a red, whereas red shows purple

krwq, 2020-04-22:
Also if you've copied the sample did you force: `var neo = new Ws2812b(spi, Count);` - the original code has it ifdef-ed with Ws2808 so the sample can share the code.

If any of the above didn't help can you share specific device you got? (i.e. link to where to get it).

Also, are you running on RPi? Which version if so?

DanielSSilva, 2020-04-22:
@krwq I've created with the ws2812b class specifically.
The neopixels I've got are [these](https://www.ebay.co.uk/itm/WS2812B-RGB-Strip-Pixels-1m-5m-WS2812-IC-5050-Individual-Addressable-5V-Light/251947626070?pageci=fe120b4b-f3d1-4d42-a08b-cc727ae4d1d8). I'm using a pi4 with either latest raspbian or Ubuntu. I'm not near it but will try to fetch that info when I can

krwq, 2020-04-23:
@DanielSSilva could you paste your /boot/config.txt content (perhaps something is overriding your spi settings later down), your csproj, .cs file and exact commands you used to build and run?

I have (almost?) identical lights in my house running on .NET all the time and it seems you might have forgotten something very simple - perhaps even physical connection.

Did you try to run same lights with Arduino to check if the lights are not defective or run sample using Python?

DanielSSilva, 2020-04-23:
I'm currently running on the rpi4 with latest raspbian.
```
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

<details>
  <summary>/boot/config.txt </summary>

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
core_freq=250
core_freq_min=250
```
</details>

Here's the .cs and csproj (both adapted from the sample code on this repo)
<details>
<summary> cs file  and csproj</summary>

cs file

``` csharp
 internal class Program
    {
        // Configure the count of pixels
        private const int Count = 1;
        public static void Main()
        {
            var settings = new SpiConnectionSettings(0, 0)
            {
                ClockFrequency = 2_400_000, Mode = SpiMode.Mode0, DataBitLength = 8
            };
            // Create a Neo Pixel x8 stick on spi 0.0
            var spi = SpiDevice.Create(settings);
            var neo = new Ws2812b(spi, Count);
            // Display basic colors for 5 sec
            BitmapImage img = neo.Image;
            img.Clear();
            img.SetPixel(0, 0, Color.Red);
            neo.Update();
        }
    }
```
The csproj

```charp
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>netcoreapp2.1</TargetFramework>
    <!--Disabling default items so samples source won't get build by the main library-->
    <EnableDefaultItems>false</EnableDefaultItems>
  </PropertyGroup>

  <ItemGroup>
    <Compile Include="*.cs" />
    <Compile Include="../Common/Iot/Device/Graphics/BitmapImage.cs" />
    <None Include="README.md" />
    <ProjectReference Include="$(MainLibraryPath)System.Device.Gpio.csproj">
      <AdditionalProperties>$(AdditionalProperties);RuntimeIdentifier=linux</AdditionalProperties>
    </ProjectReference>
  </ItemGroup>

</Project>
```
</details>

As for the commands build, it was : `dotnet publish -r linux-arm`
then I've copied the folder to the rpi and did:
` chmod +x Ws28xx.Samples` and then ran ` Ws28xx.Samples`

>Did you try to run same lights with Arduino to check if the lights are not defective or run sample using Python?

Using python, red returns as expected. Although I _think_ that the python code is not using SPI (the pin is the [pwm0](https://pinout.xyz/pinout/pin12_gpio18)

Note: On a side note, yellow is white.

krwq, 2020-04-23:
@DanielSSilva the only two things I can think of (to try in this order):
- connect your data pin here: https://pinout.xyz/pinout/pin19_gpio10# (that's SPI0's output)
- put `dtparam=spi=on` just before `core_freq=250` so that it's 3rd line before end

perhaps python implemented it differently 😄

DuncanF, 2020-06-18:
I've commented on https://github.com/dotnet/iot/issues/1064 with code that fixed the issue for my Adafruit Neopixel Ring 60/4 - this may be same problem if the LED being tested is RGBW instead of RGB.

chris03, 2020-12-03:
Same problem with my 1meter (60LEDS) Ws2812b led strip.
The `Adafruit Neopixel Ring` code from Issue https://github.com/dotnet/iot/issues/1064 didn't work either.
I used the following lib that worked correctly: https://kenssamson.github.io/rpi-ws281x-csharp/

Ellerbach, 2020-12-04:
@chris03 did you fixed the SPI clock in /boot/config.txt to 250?
Our .NET IoT implementation is using SPI. The one you pointed is using PWM and DMA with a native driver. You still can use the other driver with SPI. And you may want to compare both to check that the low level configuration is correctly implemented.

chris03, 2020-12-04:
@Ellerbach Yes, core_freq=250 and core_freq_min=250 are my last two lines in config.txt 
I tried the other lib with SPI and it works correctly.

Ellerbach, 2021-07-01:
[Triage] Closing this one as duplicate of #1328

closed 2021-07-01 by Ellerbach