dotnet/iot #1048: GPIO Pin extension

FrancoCaprio, 2020-04-21:
We are using a Raspberry PI 3 with a extended GPIO up to GPIO 44. unfortunately I would like to use the RaspberryPi3Driver but I cannot override the Pin Mapping Method or the checks. because they are private/internal or protected.

`
        protected internal override int PinCount => 28;
        private void ValidatePinNumber(int pinNumber)
        protected internal override int ConvertPinNumberToLogicalNumberingScheme(int pinNumber)
`

It should be possible to override these. Maybe I am missing out something here.

pgrawehr, 2020-04-21:
The PinCount property is `protected internal`, which means that they are protected _or_ internal. So you can overwrite it. 
However: There's a bug in ValidatePinNumber that it does not use the `PinCount` property. That should get fixed sometime.

krwq, 2020-04-21:
Thanks for fixing @pgrawehr. To use latest bits use these instructions (or nuget sources specifically):
https://github.com/dotnet/iot#net-cli

I think build should be available no later than tomorrow

PR #1049 (2020-04-21): ValidatePinCount should use the overridable member
Fixes #1048
closed 2020-04-21 by krwq