dotnet/iot #1055: Support Orange PI boards

davidjmallar, 2020-04-28:
Usually, we prefer the [Orange PI](http://www.orangepi.org/) products over the Raspberry, because they are much cheaper. Especially the [Orange PI Zero](http://www.orangepi.org/orangepizero/).
But, there is no Dotnet support yet.
That would be awesome if this library could handle at least the OPI Zero boards.

pgrawehr, 2020-04-28:
The OPi 4 could actually work, as it has very similar hardware to the Raspberry Pi 3. Have you tried installing dotnet on it? Anyway, if dotnet is not supported, this is the wrong place to ask, this Repo is only about support for the extra hardware (GPIO, I2C, etc) not about dotnet support in general.

krwq, 2020-04-28:
Possibly this will also improve it? https://github.com/dotnet/iot/pull/780

davidjmallar, 2020-05-05:
> Possibly this will also improve it? #780

Thank you. That is indeed improving it!

closed 2020-05-05 by davidjmallar