dotnet/iot #1063: Good option for a resistor array?

richlander, 2020-04-30:
This is a bit off-topic, but I'm hoping you can help me.

I recently learned about resistor network and resistor arrays, in this [video](https://www.youtube.com/watch?v=Ys2fu4NINrA&feature=youtu.be&t=834). Does anyone have a good source of these? They look super useful.

I did find [this product](https://www.amazon.com/gp/product/B07BPYR8KR/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1), which is also super useful, but isn't the same as what was in the video.

krwq, 2020-04-30:
They are 50c for 10 on aliexpress: https://www.aliexpress.com/item/33039985638.html?spm=a2g0o.productlist.0.0.56c2123dT1SSj5&algo_pvid=78d93417-0fc7-4e0e-b5ab-8bf096cf3178&algo_expid=78d93417-0fc7-4e0e-b5ab-8bf096cf3178-0&btsid=0ab6f81615882717811597009e7117&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

I bet you can find them even cheaper if you search more

richlander, 2020-04-30:
That looks like the shift register, not the resistor array.

krwq, 2020-04-30:
I noticed a second later (thought chip id in the title was what you were looking for). Perhaps digikey:
https://www.digikey.com/products/en/resistors/resistor-networks-arrays/50?utm_adgroup=Resistors&utm_source=google&utm_medium=cpc&utm_campaign=Dynamic%20Search&utm_term=&utm_content=Resistors&gclid=Cj0KCQjw7qn1BRDqARIsAKMbHDYKZv1ZvKJzNY0Iddh1vtE3sT5l29c_ps79ZBEZLgqyBDdk13-dXcIaAmFmEALw_wcB

krwq, 2020-04-30:
Note you will need to select the packaging as DIP-16 in the extra filter:
https://www.digikey.com/products/en/resistors/resistor-networks-arrays/50?k=&pkeyword=&sv=0&pv16=107879&sf=1&FV=1%7C139540%2C1%7C147533%2C-8%7C50&quantity=&ColumnSort=0&page=1&pageSize=25

Unless you go bulk you pretty much have 3 choices right now:
- https://www.digikey.com/product-detail/en/vishay-dale/MDP1603200RGE04/MDP200F-ND/1544628
- https://www.digikey.com/product-detail/en/bourns-inc/4116R-1-221/4116R-1-221-ND/158401
- https://www.digikey.com/product-detail/en/bourns-inc/4116R-1-221LF/4116R-1-221LF-ND/1088628

richlander, 2020-04-30:
Yup, I eventually found those. Thanks!!

closed 2020-04-30 by richlander