dotnet/iot #1064: WS2812b erroneous colors for certain values

mlowijs, 2020-05-02:
**Describe the bug**

Certain color values produce erroneous colors on on WS2812b LED strips.

**Steps to reproduce**

I have two LEDs in series connected to the SPI port on a Raspberry Pi 3 A+.

When looping through all the values for a color, some values turn off the LEDs or produce seemingly random colors on the LEDs, though they are not random; for each run of the code the produced colors are the same every time.

```csharp
public static async Task Main()
{
    var settings = new SpiConnectionSettings(0, 0)
    {
        ClockFrequency = 2_400_000, Mode = SpiMode.Mode0, DataBitLength = 8
    };

    var spi = SpiDevice.Create(settings);
    var neo = new Ws2812b(spi, 2);

    var image = neo.Image;
    image.Clear();

    for (var i = 0; i < 256; i++)
    {
        Console.WriteLine(i);

        image.SetPixel(0, 0, Color.FromArgb(i, i, i));
        image.SetPixel(1, 0, Color.Black);
        neo.Update();

        await Task.Delay(1000);
    }
}
```

**Expected behavior**

With the above code I would expect the first LED to go from off to full brightness white in 256 steps, while the second LED should be off all the time.

**Actual behavior**

Value of `i` | approximate color of first LED | approximate color of second LED
----------- | ------------------------------ | ---------------------------------
0 | off | off
1 | white (1, 1, 1) | off
2 | white (2, 2, 2) | off
3 | white (3, 3, 3) | off
4 | off | off
5 | pinkish | green
6 | blue | green (brighter than 5)
7 | pinkish | green (dimmer than 6)
8 | white (8,8,8) | off
9 | white (9,9,9) | off
10 | white (10,10,10) | off
11 | white (11,11,11) | off
12 | very light green | off
13 | same as 5 | same as 5
14 | same as 6 | same as 6
15 | same as 7 | same as 7
16 | white (16,16,16) | off

And this pattern repeats every few values of `i`.

**Versions used**

Add following information:

<details>
<summary>dotnet --info build machine</summary>
.NET Core SDK (reflecting any global.json):
 Version:   3.1.102
 Commit:    573d158fea

Runtime Environment:
 OS Name:     Mac OS X
 OS Version:  10.15
 OS Platform: Darwin
 RID:         osx.10.15-x64
 Base Path:   /usr/local/share/dotnet/sdk/3.1.102/

Host (useful for support):
  Version: 3.1.2
  Commit:  916b5cba26

.NET Core SDKs installed:
  2.2.207 [/usr/local/share/dotnet/sdk]
  3.0.101 [/usr/local/share/dotnet/sdk]
  3.1.102 [/usr/local/share/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.2.8 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.2.8 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.1 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.2 [/usr/local/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.2.8 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.1 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.2 [/usr/local/share/dotnet/shared/Microsoft.NETCore.App]
</details>

<details>
<summary>dotnet --info Raspberry Pi</summary>
  It was not possible to find any installed .NET Core SDKs
  Did you mean to run .NET Core SDK commands? Install a .NET Core SDK from:
      https://aka.ms/dotnet-download

Host (useful for support):
  Version: 3.1.3
  Commit:  4a9f85e9f8

.NET Core SDKs installed:
  No SDKs were found.

.NET Core runtimes installed:
  Microsoft.NETCore.App 3.1.3 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
</details>

Version of `System.Device.Gpio` package = 1.1.0-prerelease.20153.1
Version of `Iot.Device.Bindings` package = 1.1.0-prerelease.20153.1

<details>
<summary>/boot/config.txt</summary>
# See /boot/overlays/README for all available options

gpu_mem=64
initramfs initramfs-linux.img followkernel
dtparam=spi=on
core_freq=250
core_freq_min=250
</details>

pgrawehr, 2020-05-02:
Duplicate of #1042?

krwq, 2020-05-04:
@mlowijs this sounds similar, can you try things I suggested in the other issue? If this doesn't help I'm guessing we've regressed this binding at one point (although don't recall any PRs in this area)

mlowijs, 2020-05-05:
Sure, here we go:

- Double check `/boot/config.txt`: I did and it's the same as the one in the issue.
- Are they really WS2812B LEDs: I can't be sure about that. They work fine with other libraries (e.g. in Python) that use PWM/DMA instead of SPI for controlling the LEDs.
- Force `new Ws2812b()`: My code sample does that already.
- Pin connections: data pin is on 19, GND on 20 and Vcc on 4.
- Command to build: `dotnet publish -c Release -r linux-arm --no-self-contained`

<details>
<summary>.csproj</summary>
<Project Sdk="Microsoft.NET.Sdk">
    <PropertyGroup>
        <OutputType>Exe</OutputType>
        <TargetFramework>netcoreapp3.1</TargetFramework>
    </PropertyGroup>
    <ItemGroup>
      <PackageReference Include="Iot.Device.Bindings" Version="1.1.0-prerelease.20153.1" />
      <PackageReference Include="System.Device.Gpio" Version="1.1.0-prerelease.20153.1" />
    </ItemGroup>
</Project>
</details>

DuncanF, 2020-06-18:
I was getting this issue with Adafruit Neopixel Ring 60/4
I think the root cause is that Ws2812b is for RGB LED strips, but the one I have is RGBW - so the 24 bit stride means it would get colours incorrect and when trying to step through the 60 LEDs on the ring it would only address 3/4 of them (every 2 out of 4 pixels set would actually light up a pair of LED positions on the ring.)

I fixed this locally by adding my own classes and currently they seem to work okay (can address all LEDs via correct pixels, colours correct):

```
    /// <summary>
    /// Custom LED driver for NeoPixel RGBW devices
    /// Iot.Device.Ws28xx.Ws2812b doesn't support RGBW LED strips - causing light colours to be wrong
    /// and can only access 3/4 of them
    /// Adapted from https://github.com/dotnet/iot/blob/master/src/devices/Ws28xx/Ws2812B.cs
    /// </summary>
    class NeoPixel : Ws28xx
    {
        /// <summary>
        /// Constructs NeoPixel instance
        /// </summary>
        /// <param name="spiDevice">SPI device used for communication with the LED driver</param>
        /// <param name="width">Width of the screen or LED strip</param>
        /// <param name="height">Height of the screen or LED strip. Defaults to 1 (LED strip).</param>
        public NeoPixel(SpiDevice spiDevice, int width, int height = 1)
            : base(spiDevice)
        {
            Image = new BitmapImageNeo4(width, height);
        }
    }
```

```
    /// <summary>
    /// Adapted from https://github.com/dotnet/iot/blob/master/src/devices/Ws28xx/BitmapImageNeo3.cs
    /// DKF: Added 4th component to convert from 24bit RGB to 32bit RGBW values
    /// Special 32bit RGB format for Neo pixel LEDs where each bit is converted to 3 bits.
    /// A one is converted to 110, a zero is converted to 100.
    /// </summary>
    internal class BitmapImageNeo4 : BitmapImage
    {
        private const int BytesPerComponent = 3;
        private const int BytesPerPixel = BytesPerComponent * 4;
        // The Neo Pixels require a 50us delay (all zeros) after. Since Spi freq is not exactly
        // as requested 100us is used here with good practical results. 100us @ 2.4Mbps and 8bit
        // data means we have to add 30 bytes of zero padding.
        private const int ResetDelayInBytes = 30;

        public BitmapImageNeo4(int width, int height)
            : base(new byte[width * height * BytesPerPixel + ResetDelayInBytes], width, height, width * BytesPerPixel)
        {
        }

        public override void SetPixel(int x, int y, Color c)
        {
            var offset = y * Stride + x * BytesPerPixel;
            Data[offset++] = _lookup[c.G * BytesPerComponent + 0];
            Data[offset++] = _lookup[c.G * BytesPerComponent + 1];
            Data[offset++] = _lookup[c.G * BytesPerComponent + 2];
            Data[offset++] = _lookup[c.R * BytesPerComponent + 0];
            Data[offset++] = _lookup[c.R * BytesPerComponent + 1];
            Data[offset++] = _lookup[c.R * BytesPerComponent + 2];
            Data[offset++] = _lookup[c.B * BytesPerComponent + 0];
            Data[offset++] = _lookup[c.B * BytesPerComponent + 1];
            Data[offset++] = _lookup[c.B * BytesPerComponent + 2];
            // DKF added: Also set W component from Color alpha
            Data[offset++] = _lookup[c.A * BytesPerComponent + 0];
            Data[offset++] = _lookup[c.A * BytesPerComponent + 1];
            Data[offset++] = _lookup[c.A * BytesPerComponent + 2];
        }

        private static readonly byte[] _lookup = new byte[256 * BytesPerComponent];
        static BitmapImageNeo4()
        {
            for (int i = 0; i < 256; i++)
            {
                int data = 0;
                for (int j = 7; j >= 0; j--)
                {
                    data = (data << 3) | 0b100 | ((i >> j) << 1) & 2;
                }

                _lookup[i * BytesPerComponent + 0] = unchecked((byte)(data >> 16));
                _lookup[i * BytesPerComponent + 1] = unchecked((byte)(data >> 8));
                _lookup[i * BytesPerComponent + 2] = unchecked((byte)(data >> 0));
            }
        }
    }
```

Ellerbach, 2021-07-01:
[Triage] related to #1328. Up to grab!

Ellerbach, 2021-07-01:
[Triage] Closing this one as duplicate of #1328

closed 2021-07-01 by Ellerbach