dotnet/iot #1078: SHT1x support

frogcrush, 2020-05-16:
**Is your feature request related to a problem? Please describe.**

Currently there is support for the SHT3x sensors. However, there is no support for the older SHT1x sensors which are still widely available for purchase. I would like to add support for the SHT1x line of devices. 

[image]

**Describe the ideal solution**

Support the SHT1x line of sensors, either by modifying the existing SHT3x code or by using that code as a basis.

frogcrush, 2020-05-16:
Datasheet: https://www.sparkfun.com/datasheets/Sensors/SHT1x_datasheet.pdf

pgrawehr, 2020-05-16:
Probably related: #1067

frogcrush, 2020-05-18:
This is in-progress in #1080

Ellerbach, 2021-07-01:
[Triage] On PR #1080, there is most likely a rebase to do and very little efforts to move this one forward. Up to grab!

krwq, 2023-01-19:
[Triage] We're not tracking binding requests. If there is anything to discuss please re-open.

closed 2023-01-19 by krwq