dotnet/iot #1079: Relay device

frogcrush, 2020-05-16:
I would like to add support for some different relay boards. These devices are super simple and support for this would be to add consistency for code in projects more than anything. 

This device would simply keep track of what port a relay is on, and provide methods for turning it on and off by setting that pin high and low. 

[image]

pgrawehr, 2020-05-16:
Sounds good, but make it such that the number of relay channels is configurable. There are such boards with 1, 2, 4, or even 8 channels.

krwq, 2020-05-17:
I have mixed feelings on these since this will have very little logic compared to GpioController, if you guys feel like this is useful please at least add support for normally open/closed types of relays so that user doesn't have to switch

frogcrush, 2020-05-18:
@kwkraus I like that idea. I would agree that there isn't much more logic than GpioController. However, it would be more readable and also easier for entry-level people to understand.

frogcrush, 2020-05-18:
> Sounds good, but make it such that the number of relay channels is configurable. There are such boards with 1, 2, 4, or even 8 channels. 

So have it RelayBoard instead of Relay? That could make sense.

krwq, 2020-05-18:
Perhaps something along the lines of

```
RelayBoard b = new RelayBoard(numberOfRelays: 6);
var r1 = b.CreateRelay(RelayType.NormallyOpen); // this could default to normally closed
r1.On = true;
```

frogcrush, 2020-06-23:
Here is the plan for Relay:

Methods:
Relay(int pin, GpioController controller)
TurnOn()
TurnOff()
Toggle()
SetState(bool on)

Properties
bool On (get & set, calls SetState)
int Pin (get)
RelayType RelayType

And the RelayType enum:
NormallyOpen (default)
NormallyClosed

I agree RelayBoard is probably uncesessary; the only thing I could see it being useful for is setting all Relays to the same RelayType. However, I don't think it will add enough code clarity (as the Relay class will) to bother adding it.

krwq, 2020-06-28:
@frogcrush generally speaking we usually don't add redundancy to the methods, so I'd recommend to either stick with property `On` or method `SetState` or similarly named. (sorry for late response, I'm currently on vacation)

pgrawehr, 2020-06-28:
The RelayType should probably be provided in the ctor (and be read-only later).

HumJ0218, 2020-11-02:
There is another relay module that is controlled via SPI using a shift register such as 74595. Does the control of this module also need to be implemented here?

Ellerbach, 2020-11-02:
> There is another relay module that is controlled via SPI using a shift register such as 74595. Does the control of this module also need to be implemented here?

I guess in this case https://github.com/dotnet/iot/tree/master/src/devices/ShiftRegister would be your friend?

frogcrush, 2020-11-11:
> There is another relay module that is controlled via SPI using a shift register such as 74595. Does the control of this module also need to be implemented here?

Do you happen to have a link to info on one of these? I think support could be added for those; however, would too many separate types of devices then be covered by the one project?

And in this case, should we have an IRelay/IRelayBoard? (this _would_ help if you're using DI or unit testing...) and have GpioRelay/SpiRelay/I2cRelay/etc... I feel like there is a benefit to beginners if we did this, but...

HumJ0218, 2020-11-11:
I'm worried about the finished module. Previously found on Taobao some SPI relay design(using key word `继电器 SPI`), thought there will be a lot of such boards. However, when I searched for `relay SPI` on Amazon, I didn't get any results.
So these designs look more like the product of personal DIY.

But I was surprised to find that the `TLE7234SE` chip completely realized the function of using SPI to control the 8 relays. So I think it's a good choice to abstract relay devices into interfaces and then let developers and users implement specific models.

http://www.leomodulos.com.br/datasheets/Datasheet-Infineon-TLE-7234-SE.pdf

Ellerbach, 2021-07-01:
[Triage] There is an unfinish pull request #1079. This is up to grab to continue, seems like only few efforts needed.

krwq, 2023-01-19:
[Triage] We're not actively tracking binding requests. Please send a PR, if there is any discussion needed please reopen.

closed 2023-01-19 by krwq