dotnet/iot #1081: Adopt UnitsNet in Iot.Device.Bindings

krwq, 2020-05-19:
Since yesterday we are using UnitsNet package for units.

We should adopt this library further and replace as many units/doubles with the better version of the APIs.

I think it makes sense to split this work into two parts:
- investigate which type of devices will benefit from what units
- do the PRs (one per unit) - avoid merge conflicts with existing PRs though if you notice before you start working on that

Ref: https://github.com/dotnet/iot/issues/869

@shaggygi @Frankenslag @ZhangGaoxing @MarkCiliaVincenti @Ellerbach @joperezr @pgrawehr

anyone interested in helping?

pgrawehr, 2020-05-19:
I already started converting some bindings. I think it's easier to go binding by binding than unit by unit. Many bindings that currently use doubles as return values should actually return quantities instead (ADCs, Current sensors, etc)

pgrawehr, 2020-05-20:
@krwq : Question: Since changing the return type (typically from double to an unit) will be breaking for many bindings, shall I also fix typos in property/method names? As an example, I've just stepped over this (from Vl53L0X\Information.cs): 

```
        /// <summary>
        /// Get the offset in micrometers
        /// Formula from the official API
        /// </summary>
        public int OffsetMicrometers => SignalRateMeasFixed1104_400_Micrometers != 0 ? (((int)SignalRateMeasFixed1104_400_Micrometers - (400 << 4) * 1000) >> 4) * -1 : 0;

        /// <summary>
        /// Get the signal rate measurement fixed point 400 micrometers
        /// Formula from the official API
        /// </summary>
        public uint SignalRateMeasuementFixed400Micrometers => SignalRateMeasFixed1104_400_Micrometers << 9;
```
- Problem 1: The documentation is unclear (is the second property really a distance)? [No, it's a frequency]
- Problem 2: "Measuement" is obviously spelled incorrectly
- Problem 3: When changing "OffsetMicrometers" to a `Distance` entity, it does not make sense to keep the "Micrometers" part in the property name

Shall I: 
- Add a new property `Distance Offset`, keeping or replacing the old?
- The new property `Frequency SignalRateMeasurementFixed400Micrometers` shall have it's spelling fixed and the old one removed? (and then I think it should say `SignalRateMeasurementAtFixed400Micrometers` if I understand the docs correctly).

pgrawehr, 2020-05-20:
Next finding: I think (relative) humidity should use the quantity "Ratio". This has an unit "Percent", which will usually be the right one to use. This avoids problems with the range of such a number (0...1 or 0...100)?

krwq, 2020-05-20:
@pgrawehr:
- if you have a case where you got the suffix with the unit then please remove the suffix and leave only 1 property
- if it's misnamed/misdocumented please fix all cases you spot and write down somewhere and put it in the PR description. If not sure then create an issue and don't touch
- for relative humidity and other items you're not sure about, if there is a direct support for this unit then use it, if not then let's create an issue about it and solve separately

Ellerbach, 2020-09-15:
@krwq, how much efforts still needs to be done in the all up repository to fully support UnitsNet? 

I recently open few issues with bindings/helpers not supporting it: #1181, #1180.
Also PR to improve documentation on it's way: #1159

pgrawehr, 2020-09-18:
It's not so much - I've done quite a big part already, but apparently missed a few. I'll continue on this, but it's not my 1st prio right now. I'm a bit hessitant in opening yet more PRs, especially when some important issues are open (like #1145).

Ellerbach, 2020-09-18:
OK, so let's keep this issue open, let's wait for the important one to go thru and once we've done that, we can go back to fixing the few missing ones. 
We can as well open specific issues for missing ones when we catch something like I did.
And for the new ones, we then have to make sure they'll support those units.

Ellerbach, 2021-07-01:
[Triage] Seems that the work is done. If anything is missing, this will be addressed 1 by 1.

closed 2021-07-01 by Ellerbach