dotnet/iot #1082: Please extend the library usable for arm64

Smartcat168, 2020-05-20:
I tried the library on Ubuntu64 no a RPI 3B+. It did not work. 
Guess the library is only usable for arm32 only now ?

joperezr, 2020-05-20:
Thanks a lot for the issue report! It shouldn't, The library should work fine with arm64 architectures but I'm not sure if we've ever tested on Ubuntu64. Which version of the package are you using? I ask because we've made changes since we shipped 1.0.0 that might make it so that it works better on other archs. Also, do you mind explaining a bit more what do you mean by "It did not work"? Can you share exception logs or the output you got that lead you to think that it doesn't work?

krwq, 2020-05-20:
Would be useful if you've shared the sample you were trying, your csproj and how did you build so that we can check this looks ok

Smartcat168, 2020-05-21:
Hi

Thanks for replying to my post.
My Program is a simple one,  just to toggle a pin on the RPI GPIO

using System;
using System.Device.Gpio;
using System.Threading;

namespace ConsoleApplication1
{
    class Program
    {
        static void Main()
        {
            int ledPin1 = 17;
            GpioController controller = new GpioController();
            controller.OpenPin(ledPin1, PinMode.Output);

            while (true)
            {
                controller.Write(ledPin1, PinValue.High);
                Console.WriteLine("High");
                Thread.Sleep(500);
                controller.Write(ledPin1, PinValue.Low);
                Console.WriteLine("Low");
                Thread.Sleep(500);
            }
        }
    }
}

Machine ==> RPI 3B+
Machine O/S ==> 64 bit Ubuntu Server 20.04 LTS
dotnet package on RPI side ==>  dotnet-sdk-3.1.202-linux-arm64.tar.gz

This is the project file

<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.1</TargetFramework>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference
Include="runtime.linux-arm.runtime.native.System.IO.Ports" Version="4.7.0"
/>
    <PackageReference Include="System.Device.Gpio" Version="1.0.0" />
    <PackageReference Include="System.IO.Ports" Version="4.7.0" />
  </ItemGroup>

</Project>

The program is published with the following settings

Target Framework ==> netcoreapp3.1
Deployment Mode ==> Framework Dependent
Target runtime ==> Portable

This is the error I get when I try running the dll

ubuntu@ubuntu:~/publish$ *dotnet TryArm64.dll*
Unhandled exception. System.IO.IOException: Error 13 initializing the Gpio
driver.
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.Initialize()
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber, PinMode
mode)
   at ConsoleApplication1.Program.Main() in
C:\Users\smart\Desktop\DC210_Testplan\tryArm64\TryBasicConsole\Program.cs:line
20
Aborted (core dumped)

If I publish with target runtime as 'linux-arm' I will get this error

ubuntu@ubuntu:~/publish$ dotnet TryArm64.dll
Unhandled exception. System.BadImageFormatException: Could not load file or
assembly '/home/ubuntu/publish/TryArm64.dll'. An attempt was made to load a
program with an incorrect format.

krwq, 2020-05-21:
can you replace `new GpioController()` with `new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver())` and see if that fixes the issue? If seeing issue about libgpiod missing just install it.

I suspect the detection logic chose the wrong controller or there is a bug in the auto-detected controller

pgrawehr, 2020-05-21:
Error 13 means "permission denied". I can imagine that access to the GPIO ports is not allowed for everyone on Ubuntu.

@Smartcat168 : Can you try to run your application as root (with sudo)?

krwq, 2020-05-21:
I think we should add some code translation in the GPIO code (at least for the most common errors)

pgrawehr, 2020-05-22:
@krwq : I haven't had time to investigate this fully, but it looks like `Win32Exception.GetErrorMessage` already does the trick, it even has an unix implementation: https://github.com/dotnet/runtime/blob/master/src/libraries/Microsoft.Win32.Primitives/src/System/ComponentModel/Win32Exception.Unix.cs

Smartcat168, 2020-05-22:
Hi
I was able to get the gpio working in Ubuntu 64 bit after issuing the
command below.

sudo chmod 777 /dev/gpiomem

thanks

pgrawehr, 2020-05-24:
@Smartcat168 : Was that change permanent or do you have to redo this after every boot? If so, you might need to add an udev rule. That's a service that manages non-root access to virtual file system entries such as /dev/gpiomem or also /sys/class/* entries (i.e. pwm or i2c access).

Smartcat168, 2020-05-25:
Yes I need to issue command to re-enable GPIO after every boot.
Now figuring out what to put in udev to make it persistent after boot.
Thanks for the advise.

MihaMarkic, 2020-06-07:
I need that chmod as well on Raspberry PI OS. @pgrawehr can you elaborate on those udev rules?

pgrawehr, 2020-06-07:
@MihaMarkic : On Raspbian, this should not normally be necessary, as it is pre-configured for this to work (with the exception of the PWM driver, which needs an udev rule to work as non-root). 

Here's the udev rule I use for pwm: (copy to /etc/udev/rules.d/55-pwm.rules)

```
SUBSYSTEM=="pwm*", PROGRAM="/bin/sh -c '\
        chown -R root:gpio /sys/class/pwm && chmod -R 770 /sys/class/pwm;\
        chown -R root:gpio /sys/devices/platform/soc/*.pwm/pwm/pwmchip* && chmod -R 770 /sys/devices/platform/soc/*.pwm/pwm/pwmchip*\
'"
```

[55-pwm.rules.txt](https://github.com/dotnet/iot/files/4742240/55-pwm.rules.txt)

pgrawehr, 2020-06-07:
@MihaMarkic Note: Depending on the functions you use (particularly when using interrupt driven triggers) you may need to use a recent development build and not the release version, as we've done a bunch of fixes to properly support udev.

Ellerbach, 2020-09-15:
@Smartcat168, I feel like this issue is resolved with the advices from @pgrawehr. I will then close this issue. Feel free to reopen it at any time!

closed 2020-09-15 by Ellerbach