dotnet/iot #1083: Beagle Bone support for PWM

rubberduck203, 2020-05-22:
## Summary

The `System.Device.Gpio` library expects the pwm sysfs naming to conform to the mainline kernel's documentation.

https://github.com/dotnet/iot/blob/00f9da195e6c3bb296450ab88303e031ad55aff8/src/System.Device.Gpio/System/Device/Pwm/Channels/UnixPwmChannel.cs#L41-L42

> If CONFIG_SYSFS is enabled in your kernel configuration a simple sysfs
interface is provided to use the PWMs from userspace. It is exposed at
/sys/class/pwm/. Each probed PWM controller/chip will be exported as
pwmchipN, where N is the base of the PWM chip.
> ...
> The PWM channels are numbered using a per-chip index from 0 to npwm-1.
>
> When a PWM channel is exported a pwmX directory will be created in the
pwmchipN directory it is associated with, where X is the number of the
channel that was exported.

https://www.kernel.org/doc/Documentation/pwm.txt

However, the Beagle Bone's fork of the kernel does not adhere to this convention.

https://github.com/beagleboard/linux/commit/0e09cd3599153a865e87e212ffed6d485488dd4f

Instead of `/sys/class/pwm/pwmchipY/pwmX`, it uses `/sys/class/pwm/pwmchipY/pwm-Y:X`, breaking the library.

## Proposed Solution

Expose `private readonly string _channelPath` as `protected virtual string ChannelName { get; }` and derive a `BeagleBonePwmChannel` that inherits from `UnixPwmChannel`, overriding `ChannelName` to provide a working pwm solution for the BeagleBone's version of the Linux kernel.

Something like this:

```cs
internal class BeagleBonePwmChannel : UnixPwmChannel
{
    protected override string ChannelName { get => $"{ChipPath}/pwm-{Chip}:{Channel}"; }
}
``` 

The version of the kernel can be detected by reading the `/proc/version` file which returns the following information.

```txt
Linux version 4.14-100-ti-r113 (root@x1-am57xx-beagle-x15-2gb) (gcc version 6.3.0.20170516 (Debian 6.3.0-18+deb9u1)) #1 SMP PREEMPT Wed July 31 00:01:10 UTC 2019
```

There a couple of things that might be sufficient to tell us we're running on the Beagle Bone kernel here, but I'm afraid they might be fragile.  

* We could look for `ti-`, but just because it's a Texas Instruments chip doesn't mean we're running beagle bone's forked version of the kernel. 
* We could also look for `beagle`, but we have no guarantee that the Beagle Bone organization won't rename it's build server

This would at least allow the `PwmChannel.Create()` method to continue it's automatic detection of an appropriate `PwmChannel` implementation and is similar, in spirit, to how the `GpioController` finds the "best driver" to use by reading `/proc/cpuinfo`.

https://github.com/dotnet/iot/blob/00f9da195e6c3bb296450ab88303e031ad55aff8/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L382-L384

rubberduck203, 2020-06-04:
I submitted a pull request on this, but it's been radio silence on both this issue and the PR for 2 weeks. Any chance someone can chime in on this soon @joperezr @krwq? I don't like to be pushy, I know everyone's busy and I submitted this just before a holiday, but this is blocking some other work I have going on.

joperezr, 2020-06-04:
@rubberduck203 so sorry this got missed somehow, thanks a lot for your contribution, I will take a look

rubberduck203, 2020-06-04:
No worries @joperezr. I figured it just slipped through the cracks.
Thanks for the review.

PR #1084 (2020-06-11): Support PWM on Beagle Bone
Fixes #1083
closed 2020-06-11 by joperezr