dotnet/iot #1086: TLC1543 Device Binding

SaDiablo, 2020-05-25:
Hi,
I'm working on a student's science club [project](https://github.com/ACT-ZUT/AlphaBot2.NET) using [AlphaBot2](https://www.waveshare.com/wiki/AlphaBot2-Pi) robot. There are a few yet not implemented devices here.
One of them is TLC1543 - 11 channel 10 bit ADC [[Documentation]](https://www.ti.com/lit/ds/symlink/tlc1543.pdf)
In this robot it is used for converting analog reflective infrared sensor (ITR20001/T) and battery voltage

I already have a bit of binding working (fast mode 1, without using EOC pin) and it's my first time contributing to another project. How should I proceed? And also should I file rest of the missing bindings here?

pgrawehr, 2020-05-25:
There's a [template](https://github.com/dotnet/iot/blob/master/tools/templates/DeviceBindingTemplate/README.md) for how a binding could look like. Or just have a look at some of the existing bindings (i.e for an ADC it might be the ADS1115 binding). If you've got some .NET experience, you should be able to undestand the concept and general guidelines rather quickly by lookign at what's already there. 

I personally think just listing what's missing is not very useful, since that list will eventually be very long. Maybe list those that you think would be most helpful for others, too (i.e. not exotic chips), or those where you think you would be able to contribute yourself.

krwq, 2020-06-02:
As long as you implement functionality which will cover most common use cases you don't need to implement it all. As for mentioning what's missing: yes mention but not with too many details, just what might be interesting to people (i.e. if device supports both I2C/SPI but you only add 1 then mention it, if it's some implementation detail like interrupts or something then not necessarily).

Ellerbach, 2020-09-15:
@SaDiablo, have you made any progress? Do you need any help?

SaDiablo, 2020-09-15:
> @SaDiablo, have you made any progress? Do you need any help?

I need to comment out few things, add readme and few examples and then it should be ready to go. Thanks for asking :)

Ellerbach, 2020-09-15:
@SaDiablo , check as well the documentation update for couple of Do and Don't: https://github.com/dotnet/iot/tree/master/src/devices
That should help you as well!

Ellerbach, 2021-07-01:
[Triage] This has been added already. See https://github.com/dotnet/iot/tree/main/src/devices/Tlc1543

closed 2021-07-01 by Ellerbach