dotnet/iot #1089: System.Device.Gpio does not work on Mono

krwq, 2020-05-27:
When using SPI and other protocols on Mono the Windows type is being loaded and fails to load because some fields use WinRT types which don't exist on Linux.

The workaround or quick fix is to replace `new WindowsSpiDevice(...)` with `Activator.CreateInstance(Type.GetType(...), ...)` - I'm currently investigating if this is a Mono issue or if there is any better solution to fix this. It should also be fixed for other protocols not just SPI.

```
Environment.OSVersion.Platform: Unix

Unhandled Exception:
System.TypeLoadException: Could not load type of field 'System.Device.Spi.Windows10SpiDevice:_winDevice' (1)                                       due to: Could not load file or assembly 'Windows.Devices.DevicesLowLevelContract, Version=3.0.0.0, Culture=ne                                      utral, PublicKeyToken=null' or one of its dependencies.
  at gpio_test.Program.Main (System.String[] args) [0x00044] in <3861456bb5704006a45ca22f1003fbbd>:0
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeLoadException: Could not load type of field 'System.Device.Spi.                                      Windows10SpiDevice:_winDevice' (1) due to: Could not load file or assembly 'Windows.Devices.DevicesLowLevelCo                                      ntract, Version=3.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies.
  at gpio_test.Program.Main (System.String[] args) [0x00044] in <3861456bb5704006a45ca22f1003fbbd>:0
```

krwq, 2020-05-27:
Just got info from Mono people, the better way to workaround it is moving constructor to a new method marked as `[MethodImpl(MethodImplOptions.NoInlining)]`

jkoritzinsky, 2020-05-28:
As a heads up per https://github.com/dotnet/runtime/issues/35318, usage of WinRT types should move to the new C#/WinRT-based projections via the `Microsoft.Windows.SDK.NET` package when targeting .NET 5. (https://github.com/dotnet/runtime/pull/36715 removes support for the built-in projection system)

krwq, 2020-05-28:
@jkoritzinsky do you have any example of any project transitioning, I'm not sure I'm following what's the work we should do 😄

jkoritzinsky, 2020-05-28:
The way to migrate is to remove all references to winmds and `System.Runtime.WindowsRuntime` and `System.Runtime.InteropServices.WindowsRuntime` and instead just reference the `Microsoft.Windows.SDK.NET` package.

krwq, 2020-05-28:
That sounds simple enough, thanks for the heads up, I'll let you know should we have any questions

jkoritzinsky, 2020-05-28:
By the release of .NET 5 or 6, there should be a way to reference the package implicitly when targeting Windows to reduce the deployment size.

If you find any issues, I suggest reporting them at https://github.com/microsoft/cswinrt.

closed 2020-06-01 by krwq