dotnet/iot #1091: Replace references to WinRT with Microsoft.Windows.SDK.NET

krwq, 2020-05-28:
Per https://github.com/dotnet/iot/issues/1089#issuecomment-635586758 WinRT will not have a builtin runtime support and will likely stop working after 5.0 ships.

Per @jkoritzinsky:

> The way to migrate is to remove all references to winmds and System.Runtime.WindowsRuntime and System.Runtime.InteropServices.WindowsRuntime and instead just reference the Microsoft.Windows.SDK.NET package.

joperezr, 2020-08-14:
I tried doing this and found a blocker because of issue https://github.com/microsoft/CsWinRT/issues/338 since that Windows SDK package is not currently strong-named signed. Once they fix that issue, we can come back and continue this work

pgrawehr, 2020-10-02:
@joperezr : https://github.com/microsoft/CsWinRT/issues/338 is fixed. So you might want to give it another try.

closed 2020-10-20 by joperezr