dotnet/iot #1092: IndexOutOfRangeException in Iot.Device.Media.VideoDevice

HumJ0218, 2020-05-29:
I was trying to capture an image when another still using the same camera.

There was an `IndexOutOfRangeException` thrown at `src\devices\Media\VideoDevice\VideoDevice.Converter.cs:line 139`

```csharp
    public class ImageController : Controller
    {
        private static VideoConnectionSettings settings;
        private static VideoDevice device;

        public ImageController()
        {
            if (device == null)
            {
                settings = new VideoConnectionSettings(0, (1920, 1080), Iot.Device.Media.PixelFormat.YUYV);
                device = VideoDevice.Create(settings);
            }
        }

        public FileStreamResult TakePhoto() {
            var colors = VideoDevice.YuyvToRgb(device.Capture());
            var bitmap = VideoDevice.RgbToBitmap(settings.CaptureSize, colors); // exception here

            var ms = new MemoryStream();
            bitmap.Save(ms, ImageFormat.Png);
            ms.Position = 0;

            return new FileStreamResult(ms, "image/png");
        }
    }
```

@ZhangGaoxing

Ellerbach, 2020-08-21:
Have you checked that your camera actually supports YUYV capturing?

Ellerbach, 2020-09-15:
@HumJ0218 , any chance you check that the camera actually supports the mode?

closed 2022-09-30 by msftbot[bot]