dotnet/iot #1093: Request a way to keep the camera open in VideoDevice

HumJ0218, 2020-05-29:
There are many cameras that grab pictures immediately after initialization, and there will be an unstable picture situation, ex. a full-green or black frame.

How about adding a method to keep the camera open. So users can make a delay before capture a image to avoid the green frame?

@ZhangGaoxing

krwq, 2020-05-29:
I wouldn't mind if camera stayed open for whole lifetime of the class

HumJ0218, 2020-06-05:
@krwq Thank you for your explanation. My understanding is that the ABCDE five lines are inputs from a 5-32 decoder to select the current pixel line. I have this doubt because I bought a 128x64 RGB LED screen with a label `HUB75` besides interface and only have ABC three lines as line selection lines. After some research I found that these three lines are inputs of a group of shift registers. The outputs lead the pixel lines.

My question is whether there are multiple HUB75 implementations, and they are not compatible with each other.

krwq, 2020-06-09:
I'm not sure - I suspect people implement everything specially for their devices and once they get it to work they just don't care about making it compatible with other versions (unless they have both)

Ellerbach, 2020-09-15:
@HumJ0218 , did you have a chance to look at the PR #1163? 
This should solve your issue. Please let us know so we can work on the PR and merge it. Let me know if you need any help.

HumJ0218, 2020-09-17:
@Ellerbach OK. I'll try it on weekend

Ellerbach, 2021-07-01:
[Triage] Regarding the request and the PR #1163, this seems to be addressed.

closed 2021-07-01 by Ellerbach