dotnet/iot #1097: Take advantage of libgpiod get/set multiple

rubberduck203, 2020-06-04:
**Is your feature request related to a problem? Please describe.**

One of the advantages of `libgpiod` over the `sysfs` gpio interface is the ability to read and write to multiple pins simultaneously.

The current implementation does not take advantage of this.
The `GpioController` just loops through the collection of pins.

https://github.com/dotnet/iot/blob/00f9da195e6c3bb296450ab88303e031ad55aff8/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L331-L354

**Describe the ideal solution**

Add two new virtual methods to `GpioDriver` for read/write multiple and move the logic from `GpioController` into them. e.g.

```csharp
        internal virtual void Write(ReadOnlySpan<PinValuePair> pinValuePairs)
        {
             //...
        }

        internal virtual void Read(Span<PinValuePair> pinValuePairs)
        {
            //...
        }
```

Then `LibGpiodDriver` can override these methods with calls to the appropriate functions.
  - [gpiod_ctxless_get_value_multiple](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/include/gpiod.h#n137)
  - [gpiod_ctxless_set_value](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/include/gpiod.h#n177)

This allows the vast majority of drivers to continue functioning as they do today, while allowing us to take advantage of the simultaneous read/writes of libgpiod.
I believe this is a backward compatible change.

krwq, 2020-06-04:
This doesn't sound like suggestion is to add new APIs - this is rather implementation improvement.

rubberduck203, 2020-06-04:
That’s correct. I think the issue template added that label.

Ellerbach, 2021-07-01:
[Triage] Still valid request and can improve performances.

felixcollins, 2025-09-16:
This PR #2420 moves the default behaviour to GpioDriver where it can be overridden by inheriting classes. Now it should be possible to just go ahead an implement them.