dotnet/iot #1102: GPIO pin 12 on RPi 4 doesn't pull-up

NikolaTotev, 2020-06-12:
**Describe the bug**

When attempting to use pin 12 in InputPullUp PinMode, the pin does not pull up. (This has been tested with multimeter.)

**Steps to reproduce**
Just use the following code to configure the pin:

```csharp
 GpioController controller = new GpioController(PinNumberingScheme.Logical);
            Console.WriteLine("Starting button test");

            if (mode == 1)
            {
                controller.OpenPin(buttonNumber, PinMode.InputPullUp);
            }

            if (mode == 2)
            {
                controller.OpenPin(buttonNumber, PinMode.InputPullDown);
            }

            while (true)
            {
                if (controller.Read(buttonNumber) == PinValue.Low)
                {
                    Console.WriteLine("Button pressed");
                    Thread.Sleep(TimeSpan.FromSeconds(1));
                }
            }
```

**Expected behavior**
The pin (12) should be pulled up to 3.3V and when connected to the button, the code above should work. 

**Actual behavior**
The pin (12) stays at low.

**Versions used**

Add following information:

- `dotnet --info` on the machine being used to build
- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
- Version of `System.Device.Gpio` package
- Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)

Versions used
SDK:
3.1.400-preview-015151 [C:\Program Files\dotnet\sdk]
 Version of `System.Device.Gpio` package: 1.0.0

NOTE: 
This code & setup works on RPi 3 B+. 
I have tested the pin itself by toggling high & low using C# code and it functions as expected.
When using python to pull-up the pin it works as expected, that is why I am assuming its something to do with the System.Device.Gpio package.

NikolaTotev, 2020-06-12:
Hello @krwq, 

- yes, that is the correct pin
- I have never explicitly enabled PWM0 (to be completely honest though I still don't know how to do that exactly, I'm also not sure how to check that either.)
-  yes it seems to be specific to pin 12. All of the other pins I tested work (some aren't strictly GPIO pins, some are the TX/RX, I2C or SPI pins).

pgrawehr, 2020-06-14:
@NikolaTotev : Which version of the library are you using? Pull up/Pull down modes on the Pi4 only work with the current development beta, not with the officially released version.

NikolaTotev, 2020-06-14:
@pgrawehr 
As mentioned in my first comment, the versions I'm using are:

SDK:
3.1.400-preview-015151 [C:\Program Files\dotnet\sdk]
Version of System.Device.Gpio package: 1.0.0

and  all other pins I tested function correctly with Pull-Up.

I assume it's some sort of bug just with this pin as it does share  PWM 0, but I can't say for certain. 

Speaking of the beta, what are the expected release dates?

pgrawehr, 2020-06-15:
This is probably a coincidence. Package version 1.0.0 just doesn't properly handle pullups for the Pi4 (since that requires a different low-level operation than on the Pi3, which was not documented properly at the time 1.0 was released). So that might just work, depending on the startup settings. It will fail when you try to change it (i.e. set it to pulldown) on any pin. Please try again with the prerelease version, as described in the readme here: https://github.com/dotnet/iot/blob/master/README.md

NikolaTotev, 2020-06-16:
Alright, great, thanks for the advice & explanation! I'll give it a shot in a couple of days!

Ellerbach, 2020-08-26:
@NikolaTotev, did you have a chance to give it a try? Any success?

NikolaTotev, 2020-08-31:
@Ellerbach Since I posted the issue I haven't tried again, but I will be starting a new project soon and I'll make sure to test it.

krwq, 2020-09-15:
This specific issue is about RPi 4 pull up/down not working on 1.0. This scenario wasn't supported at that time and we don't have any info about this scenario not working on the current preview. I'm voting this issue should be closed. @NikolaTotev feel free to reopen if you find issue with pull up/down with the current preview or RPi3

krwq, 2020-09-15:
> at that time it didn't work

please clarify what you mean by that - feel free to open the issue for the specific problem you're seeing (or please provide a repro steps)

closed 2020-09-15 by krwq

krwq, 2020-09-15:
Note, there is an already existing issue for detection where some of the mentioned stuff doesn't work: https://github.com/dotnet/iot/issues/1145