dotnet/iot #1103: Rationalize WinRT dependencies

richlander, 2020-06-15:
I am working on a sample. I noticed that the WinMD assets are included when publishing for a Linux RID. I wouldn't expect that to happen. In addition, we're making [changing WinRT interop in .NET 5.0](https://github.com/dotnet/runtime/issues/37672).

Here's what I'd expect:

- No Windows-specific assets when publishing for Linux.
- Provide CSWinRT assets for .NET 5.0+

Demonstration of what I see:

```console
PS D:\git\iot\src\devices\Sn74hc595\samples> type .\Sn74hc595-test.csproj
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.1</TargetFramework>
    <RuntimeIdentifier>linux-arm64</RuntimeIdentifier>
    <SelfContained>false</SelfContained>
  </PropertyGroup>

  <ItemGroup>
    <ProjectReference Include="$(MainLibraryPath)System.Device.Gpio.csproj" />
    <ProjectReference Include="../Sn74hc595.csproj" />
  </ItemGroup>

</Project>
PS D:\git\iot\src\devices\Sn74hc595\samples> dir .\bin\Debug\netcoreapp3.1\linux-arm\
PS D:\git\iot\src\devices\Sn74hc595\samples> dir .\bin\Debug\netcoreapp3.1\linux-arm64\

    Directory: D:\git\iot\src\devices\Sn74hc595\samples\bin\Debug\netcoreapp3.1\linux-arm64

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           6/15/2020 10:52 AM           7680 Sn74hc595.dll
-a---           6/15/2020 10:52 AM           2504 Sn74hc595.pdb
-a---           6/15/2020 10:52 AM           6810 Sn74hc595.xml
-a---           6/15/2020  1:12 PM         135944 Sn74hc595-test
-a---           6/15/2020  1:12 PM          11216 Sn74hc595-test.deps.json
-a---           6/15/2020  1:12 PM           7680 Sn74hc595-test.dll
-a---           6/15/2020  1:12 PM           1808 Sn74hc595-test.pdb
-a---           6/15/2020  1:12 PM            212 Sn74hc595-test.runtimeconfig.dev.json
-a---           6/15/2020  1:12 PM            154 Sn74hc595-test.runtimeconfig.json
-a---           6/15/2020  1:12 PM            440 Sn74hc595-test.xml
-a---           6/15/2020 10:52 AM         102400 System.Device.Gpio.dll
-a---           6/15/2020 10:52 AM         103827 System.Device.Gpio.xml
-a---            4/5/2020  4:20 PM        1459712 UnitsNet.dll
-a---           6/14/2020  8:47 PM          30720 Windows.Devices.DevicesLowLevelContract.winmd
-a---           6/14/2020  8:47 PM          24064 Windows.Foundation.FoundationContract.winmd
-a---           6/14/2020  8:47 PM        5612032 Windows.Foundation.UniversalApiContract.winmd
```

/cc @AaronRobinsonMSFT

joperezr, 2020-06-15:
I think this is a dupe of https://github.com/dotnet/iot/issues/1091. We have already chatted with @jkoritzinsky and have been told to move to the new model for WinRT.

richlander, 2020-06-16:
Kinda. I've raised two issues here, one of which is a dupe. I'd still like to see the Windows assets not show up on Linux.

Feel free to handle the issues however you'd like.

closed 2020-10-20 by joperezr