dotnet/iot #1118: It is not possible to display custom character created via CustomCharacter on the Lcd1602

R10Rasmus, 2020-06-26:
In the Lcd1602 it is not not possible to to display custom character created via CustomCharacter. In fx. pyton RPLCD.gpio thay can be display via the write met ode. But in Lcd1602 the write met ode only support an text or a ReadOnlySpan<byte>, not the location of the custom character.

pgrawehr, 2020-06-26:
This is supported. See the example code, which has a nice example with a walking man.

R10Rasmus, 2020-06-27:
the examples is written for Hd44780. I can be written to work with Lcd1602. Can we move the example to CharacterLcd.Sample ?

pgrawehr, 2020-06-27:
The Hd44780 is the chip that drives the LCD1602 or 2004. The extended examples should work just fine with your display.

R10Rasmus, 2020-06-28:
okay thanks

closed 2020-06-28 by R10Rasmus