dotnet/iot #1119: It is not possible to scroll text right or left on the Lcd1602

R10Rasmus, 2020-06-26:
On the Lcd1602 it is not possible to scroll the text right or left of the screen.

pgrawehr, 2020-06-26:
Did you check the example? There this feature is tested. But you might not find it really useful, as it scrolls using hardware addresses, which gets some weird side effects.

R10Rasmus, 2020-06-27:
I could not find any example with side scroll, only up scroll (which works :) )

matt4321, 2020-08-17:
> I could not find any example with side scroll, only up scroll (which works :) )

Here's a sample that I have working on the RPi 4 with dotnet core 3.1

`
using Iot.Device.CharacterLcd;
using Iot.Device.Pcx857x;
using System;
using System.Device.Gpio;
using System.Device.I2c;
using System.Threading;

namespace Screen
{
    class Program
    {
        static void Main(string[] args)
        {
            var device = I2cDevice.Create(new I2cConnectionSettings(1, 0x27));
            var driver = new Pcf8574(device);
            var lcd = new Lcd1602(registerSelectPin: 0, enablePin: 2, dataPins: new int[] { 4, 5, 6, 7 }, backlightPin: 3, readWritePin: 1, controller: new GpioController(PinNumberingScheme.Logical, driver));

            Console.CancelKeyPress += (object sender, ConsoleCancelEventArgs args) =>
            {
                Console.WriteLine("Program Has Been Exited");
                lcd.Dispose();
            };

            try
            {
                Console.WriteLine("Starting Program");
                using (lcd)
                {
                    lcd.Clear();
                    lcd.SetCursorPosition(0, 0);
                    lcd.Write("Hello World");
                    while (true)
                    {
                        lcd.ShiftDisplayRight();
                        Thread.Sleep(500);
                    }
                }
            }
            catch (Exception e)
            {
                Console.WriteLine(e);
            }
        }
    }
}

`

R10Rasmus, 2020-08-18:
cool tanks :)

closed 2020-08-18 by R10Rasmus